sarxos/webcam-capture #454: Cannot download latest jar snapshot

BobS0327, 2016-04-24:
I get the following error when attempting to download jar snapshot from your website
404 - Not Found

Path /com/github/sarxos/webcam-capture/0.3.11-SNAPSHOT/webcam-capture-0.3.11-SNAPSHOT.jar not found in local storage of repository "Snapshots" [id=snapshots]

sarxos, 2016-04-24:
Hi @BobS0327 

Are you sure you are using correct link? I went to the project main page, to the Download section

https://github.com/sarxos/webcam-capture#download

And after I click on the link I see no 404 but normal download window.

[image]

BobS0327, 2016-04-24:
I went to your home page webcam-capture.sarxos.pl and clicked on the following link..

`The latest development version JAR (aka SNAPSHOT) can be downloaded** here**.
`
which generated the previous 404 message..

sarxos, 2016-04-25:
Ok, thank you! Will fix it :)