sarxos/webcam-capture #455: No Main manifest in jar file

BobS0327, 2016-04-24:
When attempting to execute your jar file...

java -jar webcam-capture-0.3.10.jar 

I get the following message...
no main manifest attribute, in webcam-capture-0.3.10.jar

I'm using Java platform 1.8, Product 1.8.0_91

sarxos, 2016-04-24:
HI @BobS0327 

You're doing it wrong. This JAR is not executable and it's not a standalone application. It won't run even if you do ```java - jar```. This project is a _library_ which means it gives you classes you need to write your own application which makes use of webcam. Please take a look at the examples for more details.

Some basic examples:

https://github.com/sarxos/webcam-capture/tree/master/webcam-capture/src/example/java

And some more sophisticated ones, to be build with Maven:

https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-examples

closed 2016-04-24 by sarxos

BobS0327, 2016-04-24:
Hello sarxox,
Yea, my bad.  I initially (incorrectly) thought it was an executable jar.
Sorry about that.
Thanx for setting me straight.
Bob