sarxos/webcam-capture #456: IpCam device not open

ramiHedfi, 2016-04-25:
When I am trying to connect to multiple camera same time, I get the error below :
Please help me
Here is the exception.

2016-04-25 07:14:59,869 ERROR [com.github.sarxos.webcam.WebcamExceptionHandler] (webcam-panel-scheduled-executor-990) Exception in thread webcam-panel-scheduled-executor-990: com.github.sarxos.webcam.WebcamException: IpCam device not open
	at com.github.sarxos.webcam.ds.ipcam.IpCamDevice.getImage(IpCamDevice.java:297) [webcam-capture-driver-ipcam-0.3.10-RC6.jar:]
	at com.github.sarxos.webcam.ds.cgt.WebcamReadImageTask.handle(WebcamReadImageTask.java:43) [webcam-capture-0.3.10-RC6.jar:]
	at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:40) [webcam-capture-0.3.10-RC6.jar:]
	at com.github.sarxos.webcam.ds.cgt.WebcamReadImageTask.getImage(WebcamReadImageTask.java:26) [webcam-capture-0.3.10-RC6.jar:]
	at com.github.sarxos.webcam.Webcam.getImage(Webcam.java:538) [webcam-capture-0.3.10-RC6.jar:]
	at com.github.sarxos.webcam.WebcamPanel$ImageUpdater.update(WebcamPanel.java:366) [webcam-capture-0.3.10-RC6.jar:]
	at com.github.sarxos.webcam.WebcamPanel$ImageUpdater.run(WebcamPanel.java:345) [webcam-capture-0.3.10-RC6.jar:]
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:471) [rt.jar:1.7.0_80]
	at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:304) [rt.jar:1.7.0_80]
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$301(ScheduledThreadPoolExecutor.java:178) [rt.jar:1.7.0_80]
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:293) [rt.jar:1.7.0_80]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1145) [rt.jar:1.7.0_80]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:615) [rt.jar:1.7.0_80]

sarxos, 2016-04-25:
Hi @ramiHedfi,

Can you please upgrade to 0.3.11 (you are using 0.3.10-RC6) to verify if this issue persist in the newest version and provide some code sample to reproduce it?

sarxos, 2016-05-06:
Hi @ramiHedfi, any progress with this? Does issue persist after you upgraded to 0.3.11?

ramiHedfi, 2016-05-17:
Hello sarxos,

Thank you for you quick reply :) and sorry for the delayed response.

I tried to upgrade to to 0.3.12 (0.3.11 I didn't found it).
I get this error :-1: 
11:57:26,191 INFO  [com.github.sarxos.webcam.WebcamDiscoveryService] (default task-2) Discovery will not run - driver IpCamDriver does not support this feature
11:57:26,373 INFO  [com.github.sarxos.webcam.ds.cgt.WebcamOpenTask] (default task-2) Opening webcam 192.168.160.195
11:57:26,375 INFO  [com.github.sarxos.webcam.WebcamDiscoveryService] (default task-2) Discovery will not run - driver IpCamDriver does not support this feature
11:57:26,579 INFO  [com.github.sarxos.webcam.Webcam] (default task-2) Disposing webcam 192.168.160.195
11:57:26,581 INFO  [stdout] (default task-2) Driver com.github.sarxos.webcam.ds.ipcam.IpCamDriver does not support buffer access

Here is my source code :+1: 
[source-code.txt](https://github.com/sarxos/webcam-capture/files/267997/source-code.txt)

Thank you in advance

sarxos, 2016-05-20:
These are not errors. As you can see all these are at ```INFO``` level and error are on ```ERROR``` level.

Your code so soooo messy and you are doing many unnecessary things. Here is your code fixed:

```java
package com.scheidtbachman.lpr.services;

import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamPanel;
import com.github.sarxos.webcam.ds.ipcam.IpCamDeviceRegistry;
import com.github.sarxos.webcam.ds.ipcam.IpCamDriver;
import com.github.sarxos.webcam.ds.ipcam.IpCamMode;

import java.io.File;
import java.io.IOException;
import java.net.MalformedURLException;
import java.nio.ByteBuffer;
import java.util.Date;

import javax.imageio.ImageIO;

public class CamTrigger {

	/**
	 * Remember to add IP camera driver JAR to the application classpath! Otherwise you will not be
	 * able to use IP camera driver features. Driver has to be set at the very beginning, before any
	 * webcam-capture related method is being invoked.
	 */

	static {
		Webcam.setDriver(new IpCamDriver());
	}

	public ByteBuffer SaveImages(String CamIP, String idLane) throws MalformedURLException, InterruptedException, IOException {

		long startTime = System.currentTimeMillis();
		int marge = 100; // milliseconds

		Webcam webcam = null;
		WebcamPanel panel = null;
		ByteBuffer res = null;
		String CAMIPP = GetIpCamera(idLane);

		IpCamDeviceRegistry.register(CAMIPP, "http://" + CAMIPP + ":9901/video.mjpeg", IpCamMode.PUSH);

		try {

			webcam = Webcam.getDefault();
			webcam.open();

			panel = new WebcamPanel(webcam);

			for (int i = 1; i < 2; i++) {
				Thread.sleep(marge);
				File file = new File(String.format("C:\\LPR\\images\\temp\\" + idLane + "\\Img01_0%d.jpg", new Date().getTime()));
				ImageIO.write(webcam.getImage(), "JPG", file);
				System.out.println(file.getAbsolutePath());
			}

			long endTime = System.currentTimeMillis();
			long totalTime = endTime - startTime;

			System.out.println("Image files captured and saved in " + totalTime + " milliseconds");

		} catch (Exception e) {
			e.printStackTrace();
			return null;
		} finally {
			IpCamDeviceRegistry.unregister(CAMIPP);
			panel.stop();
			webcam.close();
		}

		return res;
	}
}
```