sarxos/webcam-capture #457: It doesn't recognize two line license plate

uforo, 2016-05-02:
Hello,
I have been using Javaanpr libraries in my final project but i can find a way  to customize it so as it can  recognize two lined number plates,
please help

sarxos, 2016-05-02:
Can you please be more specific? Do you have some problems with Webcam Capture API or rather with JavaANPR itself?

uforo, 2016-05-02:
Most of my Country's license plates are two lined(2 rows) so the javaANPR
fails to recognize the plate, need assistance on how to modify the matrices
so as it can recognize these plates

uforo, 2016-05-02:
JavaANPR itself

sarxos, 2016-05-02:
In that case I suggest you to open new ticket in JavaANPR project (https://github.com/oskopek/javaanpr) since this one here, where we are discussing, is Webcam Capture API which is not related to licence plates at all.

Actually this is the first time I heard about JavaANPR :)

uforo, 2016-05-02:
Thanks very much, i will keep in touch i if i will enconter some
difficulties.
Well i found it from sourceforge site.

sarxos, 2016-05-02:
Ok, take care!

closed 2016-05-02 by sarxos