sarxos/webcam-capture #458: keep open webcam during the entire application

julianlujan1, 2016-05-05:
Good night, I have been using their API which has been very good, only have a concern, as you could keep open the web cam during all my application? This with the purpose of not be open in each form (since you can use it in several forms) because this takes approximately 5 seconds to open the web cam. Can be done only once and keep it open to all forms? Many thank you in advance

sarxos, 2016-05-05:
Hi @julianlujan1,

If your webcam is UVC device (the once which is connected via USB port), then yes, you can open it when your application starts and then just reference it in all your application's forms. By doing this you don't have to open it every time. The only drawback here is the fact that webcam LED will be constantly turned on (this LED cannot be turned off).

If your webcam is IP camera, then it may be hard to keep it open for a very long time since it uses HTTP connection which (depending on IP camera server settings) may finally timeout.

I hope this answers your question.

julianlujan1, 2016-05-05:
Thank you for your prompt response. It is a usb camera, or it could be that is integrated into a laptop. This being the case, as would be the code on the first form that I want to open the camera and as serious the code in the following forms to use the webcam already open in another form. I hope you can help me. Thanks a lot.

sarxos, 2016-05-06:
You are asking how to reference objects from a different class? This is very simple - just open your webcam immediately after your application is started (in the ```main(String[] args)```) and store in some static field which you later access from different forms.

```java
public class MainApplication {
  public static Webcam webcam;
  public static void main(String[] args) {
    webcam = Webcam.getDefault();
    webcam.open();
    // other stuffs
  }
}
```

And in your form:

```java
public class SomeForm {
  void doSomething() {
    Webcam webcam = MainApplication.webcam;
    ByufferedImage image = webcam.getImage();
    // do something with image
  }
}
```

julianlujan1, 2016-05-06:
Thanks, already implement it and effectively does what I wanted, but now comes a problem, when I open a form the camera works well, open another that also uses the camera and works well, I open another form that uses the camera and is checked the camera, such as slow, and every time I open a form that uses the camera each time gets slower , until that last it does not show as if you were capturing image, it remains checked at all. Thank you in advance for your great help.

sarxos, 2016-05-07:
HI @julianlujan1,

Question:

1. Is this happening only when you open several forms at once? Or, 
2. Is this rather an issue where you open, and then close form, and then open and close another, repeat this again and again, and after such subsequent open-close operations the rendering is slow?

In case of 1 you can mitigate this issue by opening camera in asynchronous mode. In asynchronous mode all calls to ```getImage(..)``` are not blocking and thus should not decrease effective camera FPS. This can be done by passing ```true``` into the ```open(..)``` method:

```java
webcam = Webcam.getDefault();
webcam.open(true); // true = asynchronous mode, false = blocking
```

By default if you call ```webcam.open()``` it is open in blocking mode ```webcam.open(false)```.

But if your problem is a case described in 2, then it may be a bug in your code or in webcam API itself. In this case I may need some more information.

I suppose you are using ```WebcamPanel``` to render webcam video feed?

julianlujan1, 2016-05-08:
Many many thanks to true to open the camera was solved my problem, thank you very much for their immense and successful cooperation. just finished the project the upload, any doubt go back and write them thank you from colombia

sarxos, 2016-05-08:
Awesome :) Take care!

closed 2016-05-08 by sarxos