sarxos/webcam-capture #459: Is it possible to play the stream on android?

Horkin93, 2016-05-06:
I stream video from webcam, as shown in this example: https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-examples/webcam-capture-live-streaming. Is it possible to play the stream on android? If so, how this can be done? Sorry for bad english.

sarxos, 2016-05-06:
Hi @Horkin93,

The example you are pointing to is using [h.264 encoder](https://github.com/sarxos/webcam-capture/blob/master/webcam-capture-examples/webcam-capture-live-streaming/src/main/java/us/sosia/video/stream/handler/H264StreamEncoder.java) build on top of [Xuggler](https://github.com/artclarke/xuggle-xuggler). On Android side you could try using [jcodec](https://github.com/jcodec/jcodec) to decode it (if stream is a basic h.264 profile), but I will be frankly with you - I'm completely unfamiliar with video encoding/decoding on Android and have node idea how to do it. However, jcodec project is something I would start my research from.

sandeepparek, 2016-09-22:
do notify us here if you have done with that research.

alexmao86, 2016-09-22:
I think there is chance to integrate android in. I really like the idea the OOA design of this framework. great job