sarxos/webcam-capture #460: Handling an IP Camera being unplugged.

fmtsmith, 2016-05-06:
Hi,
    Thanks for your great work, it is most useful for me. I have created an app that can display up to 16 ip cameras at once. However, when one or more of the cameras get unplugged, my app hangs for quite a while, depending on how many are unplugged. I have found it hangs where a check to see if the camera is online using IpCamDevice isOnline(). Is there a way to know if a camera is unplugged and handle it more quickly?

sarxos, 2016-05-07:
Hi @fmtsmith,

I suppose this is because http timeout is too big but in older releases there is posibility to change it. I just modified the code to add this possibility and set it globally or per device. The newest JAR snapshot for the IP camera driver can be found [here](https://oss.sonatype.org/service/local/artifact/maven/redirect?r=snapshots&g=com.github.sarxos&a=webcam-capture-driver-ipcam&v=0.3.12-SNAPSHOT).

Set global timeout (**before** you set driver):

```java
System.setProperty(IpCamHttpClient.SO_TIMEOUT, "5000"); // milliseconds
System.setProperty(IpCamHttpClient.SO_CONNECTION_TIMEOUT, "5000"); // milliseconds
```

Or global timeout via command line:

```plain
$ java -Dhttp.socket.timeout=5000 -Dhttp.connection.timeout=5000 ...
```

Or in runtime:

```java
import org.apache.http.params.HttpConnectionParams;
// ...
for (IpCamDevice device : IpCamDeviceRegistry.getIpCameras()) {
  HttpConnectionParams.setSoTimeout(device.getClient().getParams(), 5000); 
  HttpConnectionParams.setConnectionTimeout(device.getClient().getParams(), 5000); 
}
```

Unfortunately I currently have no access to the IP cameras which I can disconnect so I was unable to test this code. Can you please verify if this is working? You can also play with the code of ```IpCamDevice``` on your own to introduce some other timeout mechanisms.

sarxos, 2016-05-07:
Just FYI, this is a commit introducing this possibility: https://github.com/sarxos/webcam-capture/commit/d84765baf15f308129d1534af99aa7e5c81bd858

fmtsmith, 2016-05-09:
Thank you very much for your quick response. I will test it today and let you know the results.

sarxos, 2016-05-14:
Hi, any progress with test?

fmtsmith, 2016-05-16:
Hi, sorry for the delay, we are on a deadline. Yes it worked. I set the timeout to 5 seconds, that should be good. I do have one other question if you don't mind. When I am viewing all 16 cameras at once, our application grows to about 2GB and then levels out. Do you have any tips or suggestions on how we can limit the growth of our application? Thank you so much.
Regards,
Fran