sarxos/webcam-capture #463: Connect camera wifi API

julianlujan1, 2016-05-14:
Hi, I would like to know how to connect java using its API with a camera that is connected via wifi. I could tell if it is possible and as a serious way of doing this? Thanks a lot

sarxos, 2016-05-14:
If it's connected via wifi then it's IP camera, am I correct? In this case simply use IP camera driver:

https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers/driver-ipcam

sarxos, 2016-05-14:
Ah, I forget to mention, your camera has to expose MJPEG stream to make use of IP camera driver. In current state it's unable to decode h.264 streams.

sarxos, 2016-05-14:
Here are some examples using IP camera driver functionalities: 

https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers/driver-ipcam/src/examples/java

julianlujan1, 2016-05-19:
Hello, I know how I would record my driver of one camera wifi sjcam with reference sj4000, would like to know to be able to implement the video capture with Web Cam API. Thanks a lot

sarxos, 2016-05-20:
In regards to SJ4000, it seems like there is no support for MJPEG streaming, thus there is no way to access it using Webcam Capture API.

http://s1183.photobucket.com/user/mnemennth/media/SJ4000%20Action%20Camera%20Tech%20Teardown/DSC09473.jpg.html

http://s1183.photobucket.com/user/mnemennth/media/SJ4000%20Action%20Camera%20Tech%20Teardown/DSC09472.jpg.html

mzjuin, 2016-11-17:
I use an app to switch my phone as IP cam. I obtain its IP address. but encounter some problem when pull and open in eclipse.
``
import java.net.MalformedURLException;

import javax.swing.JFrame;

import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamPanel;
import com.github.sarxos.webcam.ds.ipcam.IpCamDevice;
import com.github.sarxos.webcam.ds.ipcam.IpCamDeviceRegistry;
import com.github.sarxos.webcam.ds.ipcam.IpCamDriver;
import com.github.sarxos.webcam.ds.ipcam.IpCamMode;

public class DetectMotionFromIpCamerasExample {
static {
    Webcam.setDriver(new IpCamDriver());
}

public static void main(String[] args) throws MalformedURLException {
    IpCamDeviceRegistry.register(new IpCamDevice("LGG3", "http://192.168.0.156:8080/video", IpCamMode.PULL));
    JFrame f = new JFrame("MYRoom");
    f.add(new WebcamPanel(Webcam.getDefault()));
    f.pack();
    f.setVisible(true);
    f.setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);
}
}