sarxos/webcam-capture #464: Streaming for Network

RedToor, 2016-05-16:
I want do a remote capture cam but I not found example, thanks

sarxos, 2016-05-16:
If your remote camera is an IP camera:

https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers/driver-ipcam/src/examples/java

RedToor, 2016-05-16:
but what is the client and server?

djabereddinechabou, 2016-05-16:
so u can   can see this project
 https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-examples/webcam-capture-live-streaming it will be very helpfull i tested it on my pc and another  pc and it work very very well

RedToor, 2016-05-17:
ese proyecto usa librerias portatiles?

RedToor, 2016-05-20:
solve the problem by taking screenshots and sending them by socket directly, thank you very much for the comments.

sarxos, 2016-05-31:
Cool! I'm closing this issue.

closed 2016-05-31 by sarxos