sarxos/webcam-capture #465: setCustomViewSizes

lcssanches, 2016-05-20:
I have a LifeCam web CAM HD720 and trying to set this resolution, but no success, although this is the "max" resolution available (i'm sure because I tried with Yawcam).

The warning is this "Different size obtained vs requested - [1280x720] vs [640x480]. Setting correct one. New size is [640x480]"

Do I have to use another driver?

`webcam.setCustomViewSizes(new Dimension[]{d});
webcam.setViewSize(d);
webcam.open(true);`

Where "d" is "WebcamResolution.HD720.getSize()"

sarxos, 2016-05-20:
Hi @lcssanches,

Your code is absolutely correct. Default driver natives from OpenIMAJ will print some useful information when switched to debug mode. For this to happen you need to set ```OPENIMAJ_GRABBER_VERBOSE``` environment variable to true:

```plain
C:\>set OPENIMAJ_GRABBER_VERBOSE=true
C:\>java -cp [jars] [main-class]
```

In Eclipse, Netbeans or other IDE you can set it in runner/launcher configuration. After you run your program you should see something like:

```plain
SETUP: Setting up device 1
SETUP: USB Video Device
SETUP: Couldn't find preview pin using SmartTee
SETUP: Default Format is set to 640 by 480 
SETUP: trying format RGB24 @ 640 by 480
SETUP: trying format RGB32 @ 640 by 480
SETUP: trying format RGB555 @ 640 by 480
SETUP: trying format RGB565 @ 640 by 480
SETUP: trying format YUY2 @ 640 by 480
SETUP: Capture callback set
```

This could give us some more details why it is unable to select 1280x720.

lcssanches, 2016-05-20:
Hi @sarxos !

Thanks for the quick reply!

Well, this is the output

```
[AWT-EventQueue-0] INFO com.github.sarxos.webcam.Webcam - WebcamDefaultDriver capture driver will be used
[webcam-discovery-service] DEBUG com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver - Searching devices
[webcam-discovery-service] DEBUG com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver - Found device Microsoft LifeCam Cinema 0
OK!
[AWT-EventQueue-0] DEBUG com.github.sarxos.webcam.Webcam - Setting new resolution 1280x720
Abrindo webcam... 
[AWT-EventQueue-0] DEBUG com.github.sarxos.webcam.WebcamLock - Lock Webcam Microsoft LifeCam Cinema 0
[atomic-processor-1] INFO com.github.sarxos.webcam.ds.cgt.WebcamOpenTask - Opening webcam Microsoft LifeCam Cinema 0
[atomic-processor-1] DEBUG com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Opening webcam device Microsoft LifeCam Cinema 0
[atomic-processor-1] DEBUG com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Webcam device 0 starting session, size java.awt.Dimension[width=1280,height=720]
[atomic-processor-1] DEBUG com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Webcam device session started
[atomic-processor-1] WARN com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Different size obtained vs requested - [1280x720] vs [640x480]. Setting correct one. New size is [640x480]
[atomic-processor-1] DEBUG com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Clear memory buffer
[atomic-processor-1] DEBUG com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Webcam device com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice@63920fcc is now open
[AWT-EventQueue-0] DEBUG com.github.sarxos.webcam.Webcam - Webcam is now open Microsoft LifeCam Cinema 0
[frames-refresher-[0]] TRACE com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Next frame
[frames-refresher-[0]] TRACE com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Next frame
[atomic-processor-1] TRACE com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Webcam grabber get image pointer
[atomic-processor-1] TRACE com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Webcam device get buffer, read 921600 bytes
[AWT-EventQueue-0] DEBUG com.github.sarxos.webcam.WebcamUpdater - Webcam updater has been started
OK!
Recuperando imagem da webcam... 
[atomic-processor-1] TRACE com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Webcam grabber get image pointer
[atomic-processor-1] TRACE com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Webcam device get buffer, read 921600 bytes
[frames-refresher-[0]] TRACE com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Next frame
OK!
[AWT-EventQueue-0] DEBUG com.github.sarxos.webcam.Webcam - Closing webcam Microsoft LifeCam Cinema 0
[atomic-processor-1] TRACE com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Webcam grabber get image pointer
[frames-refresher-[0]] TRACE com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Next frame
[atomic-processor-1] TRACE com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Webcam device get buffer, read 921600 bytes
[atomic-processor-1] INFO com.github.sarxos.webcam.ds.cgt.WebcamCloseTask - Closing Microsoft LifeCam Cinema 0
[atomic-processor-1] DEBUG com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Closing webcam device
[AWT-EventQueue-0] DEBUG com.github.sarxos.webcam.WebcamUpdater - Webcam updater has been stopped
[AWT-EventQueue-0] DEBUG com.github.sarxos.webcam.WebcamLock - Unlock Webcam Microsoft LifeCam Cinema 0
[webcam-lock-[Microsoft LifeCam Cinema 0]] DEBUG com.github.sarxos.webcam.WebcamLock - Lock updater has been interrupted
[AWT-EventQueue-0] DEBUG com.github.sarxos.webcam.Webcam - Webcam Microsoft LifeCam Cinema 0 has been closed
[webcam-discovery-service] DEBUG com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver - Searching devices
[webcam-discovery-service] DEBUG com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver - Found device Microsoft LifeCam Cinema 0

***** VIDEOINPUT LIBRARY - 0.1995 - TFW07 *****

VIDEOINPUT SPY MODE!

SETUP: Looking For Capture Devices
SETUP: 0) Microsoft LifeCam Cinema 
SETUP: 1 Device(s) found

***** VIDEOINPUT LIBRARY - 0.1995 - TFW07 *****

VIDEOINPUT SPY MODE!

SETUP: Looking For Capture Devices
SETUP: 0) Microsoft LifeCam Cinema 
SETUP: 1 Device(s) found

VIDEOINPUT SPY MODE!

SETUP: Looking For Capture Devices
SETUP: 0) Microsoft LifeCam Cinema 
SETUP: 1 Device(s) found

SETUP: Setting up device 0
SETUP: Microsoft LifeCam Cinema
[... 31 lines cut ...]
SETUP: filter removed NullRenderer  
SETUP: removing filter Sample Grabber...
SETUP: filter removed Sample Grabber  
SETUP: removing filter AVI Decompressor...
SETUP: filter removed AVI Decompressor  
SETUP: removing filter Smart Tee...
SETUP: filter removed Smart Tee  
SETUP: removing filter Microsoft LifeCam Cinema...
SETUP: filter removed Microsoft LifeCam Cinema  
SETUP: freeing Capture Graph 
SETUP: freeing Main Graph 
SETUP: Device 0 disconnected and freed

VIDEOINPUT SPY MODE!

SETUP: Looking For Capture Devices
SETUP: 0) Microsoft LifeCam Cinema 
SETUP: 1 Device(s) found

```

Here's the setup screen from Yawcam.
[http://s32.postimg.org/iaqnl683p/Yawcam.png](http://s32.postimg.org/iaqnl683p/Yawcam.png)

Looks like the closest found by the grabber.

frankpapenmeier, 2016-05-22:
Hi, I also had some issues with the default driver used by this library and wrote a new driver that uses the Microsoft Media Foundation in the backend. You can download it here: https://github.com/frankpapenmeier/webcam-capture-driver-native
Using this driver, you should be able to capture HD images.