sarxos/webcam-capture #467: webcal video Flickering with SWT_AWT bridge under Windows

fbuloup, 2016-05-21:
Hello !

I have read this issue : https://github.com/sarxos/webcam-capture/issues/91

I think I am trying to do the same thing and get also flickering in AWT canvas under Windows.
With exactly the same code under MacOSX, there is absolutely no flickering !

Did you solve the problem under Windows ? Someone have a solution ?

Here is my little code :

`Composite composite = new Composite(parent, SWT.EMBEDDED | SWT.BORDER);`
`Frame frame = SWT_AWT.new_Frame(composite);`
`		canvas = new Canvas() {`
`			public void paint(Graphics g) {`
`				if(image != null)`
`				g.drawImage(image, 0, 0, canvas.getWidth(), canvas.getHeight(), 0, 0, image.getWidth(), image.getHeight(), null);`
`			}`
`		};`

Windows 7 - 64 bits - Logitech HD C920

Thanks.

fbuloup, 2016-05-22:
i've found a workaround in this thread :

https://bugs.eclipse.org/bugs/show_bug.cgi?id=91157

The solution that worked for me was to override `update()` method from Canvas AWT in order to avoid background repaint (comment number 4 in previous link)

sarxos, 2016-05-22:
Hi @fbuloup,

So just for the further reference, the solution is to override ```update()``` instead of ```paint()```. Am I correct?

```java
Composite composite = new Composite(parent, SWT.EMBEDDED | SWT.BORDER);
Frame frame = SWT_AWT.new_Frame(composite);
canvas = new Canvas() {
    public void update(Graphics g) {
        if(image != null) {
            g.drawImage(image, 0, 0, canvas.getWidth(), canvas.getHeight(), 0, 0, image.getWidth(), image.getHeight(), null);
        }
    }
};

fbuloup, 2016-05-22:
Hello Sarxos,

from my experience, if you simply override `update()`, this will solve a big part but not the whole problem. Here is an abstract of the javadoc method from AWT Canvas class :

**_Updates this canvas. 
This method is called in response to a call to repaint. The canvas is first cleared by filling it with the background color, and then completely redrawn by calling this canvas's paint method. Note: applications that override this method should either call super.update(g) or incorporate the functionality described above into their own code._**

Note that `update()` calls `paint()`

Main problem comes from the fact that the canvas is cleared before calling `paint()` method which implies flickering. The solution is simply to override this method to avoid this clearing by only calling the `paint()` method. If you don't implement `paint()` method, you will have flickering when canvas size changes, which can happen when main window is resized for instance. Moreover If you want to avoid all flickering, you must also set `sun.awt.noerasebackground` property to true (see http://www.eclipse.org/articles/article.php?file=Article-Swing-SWT-Integration/index.html, Reducing Flicker section). Please note also that these problems only appear on windows platform and that implementing these workarounds don't affect behaviour on mac (i have not made any test on Linux OS).

Here is my code at the end :

`// Make sure sun.awt.noerasebackground property has been set to true in activator`
`		composite = new Composite(parent, SWT.EMBEDDED | SWT.BORDER | ``SWT.NO_BACKGROUND);`
`		Frame frame = SWT_AWT.new_Frame(composite);`
`		canvas = new Canvas() {`
`			@Override`
`			public void paint(Graphics g) {`
`				if(image != null) g.drawImage(image, 0, 0, canvas.getWidth(), ``canvas.getHeight(), 0, 0, image.getWidth(), image.getHeight(), null);`
`				else {`
`					g.setColor(Color.BLACK);`
`					g.fillRect(0, 0, canvas.getWidth(), canvas.getHeight());`
`				}`
`			}`
`			// Override update to avoid background flickering`
`			@Override`
`			public void update(Graphics g) {`
`				paint(g);`
`			}`
`		};`
`		frame.add(canvas);`

Hope it's clear without any flickering !

sarxos, 2016-05-23:
Hi @fbuloup, 

Thank you for a detailed explanation :) I will mark this issue as closed.

closed 2016-05-23 by sarxos