sarxos/webcam-capture #468: Wrong JPEG library version: library is 80, caller expects 70

ertong, 2016-05-26:
When the library is used in ubuntu 14.04 with javafx, javafx stops loading jpeg images with error:
`Wrong JPEG library version: library is 80, caller expects 70`

To reproduce this issue it is enough to run the following code 
[mavenproject4.TAR.GZ](https://github.com/sarxos/webcam-capture/files/284687/mavenproject4.TAR.GZ)
using OracleJDK8u92 32bit in ubuntu live cd.

The result should look similar to
[image]

ertong, 2016-05-26:
I suspect the issue is linked to the OpenIMAJ library. So I posted it in their tracker too: https://github.com/openimaj/openimaj/issues/105

ertong, 2016-05-27:
It looks like if we change the initialization sequence, it starts working
```
public class Main  extends Application
{    
    public static void main(String[] args) {
        Application.launch(Main.class, args);
    }
    @Override
    public void start(Stage stage) throws Exception {
        System.out.println("Webcam: "+Webcam.getDefault());
        {
            File f = new File("1.jpg").getAbsoluteFile();
            System.out.println(f.toURI().toString());
            Image image = new Image(f.toURI().toString());
            System.out.println("Image: "+image+" "+image.getWidth()+"x"+image.getHeight()+" Error:"+image.isError()+" "+image.getException());

            Group r = new Group(new ImageView(image));
            Scene s = new Scene(r);
            stage.setScene(s);
        }
        System.out.println("Webcam: "+Webcam.getDefault());
        stage.show();
    }
}
```

It can help as a workaround.

sarxos, 2016-05-31:
I described this w/a in

https://github.com/sarxos/webcam-capture/wiki/Wrong-JPEG-library-version:-library-is-80,-caller-expects-70

It can be found on the "Known Problems" list on the wiki main:

https://github.com/sarxos/webcam-capture/wiki

closed 2016-05-31 by sarxos