sarxos/webcam-capture #470: Failed to initialize BridJ

Spenhouet, 2016-05-29:
I'm trying to get [this project](https://bitbucket.org/Temdegon/greenhouse) running on my Raspberry Pi 3.
It uses this webcam-capture library in version 0.3.10 over maven.

**When i startup the project i get this exception:**

> OpenJDK Zero VM warning: You have loaded library /tmp/BridJExtractedLibraries219873062138673759/bridj.so which might have disabled stack guard. The VM will try to fix the stack guard now.
> It's highly recommended that you fix the library with 'execstack -c <libfile>', or link it with '-z noexecstack'.
> java.lang.RuntimeException: Failed to initialize BridJ
>     at org.bridj.Platform.initLibrary(Platform.java:370)
>     at org.bridj.Platform.<clinit>(Platform.java:175)
>     at com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.<clinit>(OpenIMAJGrabber.java:59)
>     at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.handle(WebcamDefaultDriver.java:56)
>     at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
>     at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1145)
>     at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:615)
>     at java.lang.Thread.run(Thread.java:745)
> Caused by: java.lang.UnsatisfiedLinkError: /tmp/BridJExtractedLibraries219873062138673759/bridj.so: /tmp/BridJExtractedLibraries219873062138673759/bridj.so: Kann die Shared-Object-Datei nicht öffnen: Datei oder Verzeichnis nicht gefunden
>     at java.lang.ClassLoader$NativeLibrary.load(Native Method)
>     at java.lang.ClassLoader.loadLibrary1(ClassLoader.java:1965)
>     at java.lang.ClassLoader.loadLibrary0(ClassLoader.java:1890)
>     at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1851)
>     at java.lang.Runtime.load0(Runtime.java:795)
>     at java.lang.System.load(System.java:1062)
>     at org.bridj.Platform.initLibrary(Platform.java:347)
>     ... 7 more
> Exception in thread "main" com.github.sarxos.webcam.WebcamException: java.util.concurrent.ExecutionException: com.github.sarxos.webcam.WebcamException: Cannot execute task
>     at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:124)
>     at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:816)
>     at com.github.sarxos.webcam.Webcam.getDefault(Webcam.java:879)
>     at com.github.sarxos.webcam.Webcam.getDefault(Webcam.java:856)
>     at com.github.sarxos.webcam.Webcam.getDefault(Webcam.java:834)
>     at com.epam.llpd.greenhouse.PiHardwareController.initCam(PiHardwareController.java:150)
>     at com.epam.llpd.greenhouse.PiHardwareController.<init>(PiHardwareController.java:53)
>     at com.epam.llpd.greenhouse.App.createGrowBox(App.java:13)
>     at com.epam.llpd.greenhouse.web.ServerStarter.createGrowBoxModel(ServerStarter.java:59)
>     at com.epam.llpd.greenhouse.web.ServerStarter.createHandlers(ServerStarter.java:64)
>     at com.epam.llpd.greenhouse.web.ServerStarter.runServer(ServerStarter.java:39)
>     at com.epam.llpd.greenhouse.web.ServerStarter.main(ServerStarter.java:33)
> Caused by: java.util.concurrent.ExecutionException: com.github.sarxos.webcam.WebcamException: Cannot execute task
>     at java.util.concurrent.FutureTask.report(FutureTask.java:122)
>     at java.util.concurrent.FutureTask.get(FutureTask.java:188)
>     at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:116)
>     ... 11 more
> Caused by: com.github.sarxos.webcam.WebcamException: Cannot execute task
>     at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.process(WebcamProcessor.java:72)
>     at com.github.sarxos.webcam.WebcamProcessor.process(WebcamProcessor.java:140)
>     at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:46)
>     at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.newGrabber(WebcamDefaultDriver.java:46)
>     at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver.getDevices(WebcamDefaultDriver.java:118)
>     at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:36)
>     at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:26)
>     at java.util.concurrent.FutureTask.run(FutureTask.java:262)
>     at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1145)
>     at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:615)
>     at java.lang.Thread.run(Thread.java:745)
> Caused by: java.lang.UnsatisfiedLinkError: org.bridj.Platform.sizeOf_ptrdiff_t()I
>     at org.bridj.Platform.sizeOf_ptrdiff_t(Native Method)
>     at org.bridj.Platform.<clinit>(Platform.java:179)
>     at com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.<clinit>(OpenIMAJGrabber.java:59)
>     at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.handle(WebcamDefaultDriver.java:56)
>     at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
>     ... 3 more

It seems to come from this library so i thought i'd ask here for help.

_I also wrote an issue for the project [here](https://bitbucket.org/Temdegon/greenhouse/issues/2/how-to-run) but that is more in gernerall on how to get the project running._

Spenhouet, 2016-05-30:
Okay i solved it.
I changed the webcam-capture version to 0.3.10 and added 

>     <dependency>
>       <groupId>com.nativelibs4java</groupId>
>       <artifactId>bridj</artifactId>
>       <version>0.7.0</version>
>     </dependency>
>     <dependency>
>       <groupId>com.github.sarxos</groupId>
>       <artifactId>webcam-capture-driver-v4l4j</artifactId>
>       <version>0.3.11</version>
>     </dependency>

as well as

>     static {
>         Webcam.setDriver(new V4l4jDriver()); // this is important
>     }

everything as decriped [here](https://github.com/sarxos/webcam-capture/wiki/How-To-Configure-Raspberry-Pi#work-with-classic-uvc-camera)

closed 2016-05-30 by Spenhouet

sarxos, 2016-05-31:
Cool :) I'm glad you fixed it.

danielbathke, 2016-07-28:
@Spenhouet, thanks for you question and answer. I run into the same problem, and it fixes.