sarxos/webcam-capture #471: Strange color displace in pictures

Spenhouet, 2016-05-30:
If i make a picture the colors are strange. Black and white ist working but blue and red ist displaced and doubles. Really strange.

Example picture with a black band, a yellow front part on a white table: [image]

Im using:

- webcam-capture 0.3.10
- webcam-capture-driver-v4l4j 0.3.11
- bridj 0.7.0

on a Raspberry Pi 3 with this [webcam](https://www.amazon.de/CSL-Mikrofon-Lichtsensor-interpoliert-Universal/dp/B00BPBQ2DM/ref=sr_1_1?ie=UTF8&qid=1464191236&sr=8-1&keywords=CSL+-+USB+Webcam+inkl.+Mikrofon+mit+6+LEDs+HD+%2B+Lichtsensor+|+%28interpoliert+12+MP%29).
With fswebcam the pictures are normal.

**What can this be and how to solve it?**

sarxos, 2016-05-31:
Hi @Spenhouet,

This doesn't look like software issue. I suppose hardware is to blame.

Some webcams has power consumption level higher than RasPi USB can deliver. This may result in strange artifacts in the image (black or dark image, strange colours, weird noise, etc). To verify this you may try to supply it with extra current.

Here is a list of webcams tested on RasPi (not entirely reliable though):

http://elinux.org/RPi_USB_Webcams

You may also try another webcam.

sarxos, 2016-05-31:
Also I think the reason why ```fswebcam``` from CLI works well is that it takes only a single image and here, even if you take only one image with Webcam Capture, the webcam is constantly refreshed in the background.

I have one more idea - please test this and save image as JPG and not PNG and do not process your image in any way (do not put date on it or change the raster in any other way), just save it. If resultant JPG image has strange colours then I would really suppose the power issue. If colours will be correct, then it is most likely caused by a different image type (non-alpha vs alpha).

```java
Webcam w = Webcam.getDefault();
w.open();
ImageIO.write(w.getImage(), "JPG", new File("test1.jpg")); // test this image
ImageIO.write(w.getImage(), "JPG", new File("test2.jpg")); // test this image
ImageIO.write(w.getImage(), "JPG", new File("test3.jpg")); // test this image
w.close();
```

Spenhouet, 2016-05-31:
Hi @sarxos,

thank you for your fast reply. I tested you example code with not addition and got 3. times the same picture:

test1.jpg:
[image]
test2.jpg:
[image]
test3.jpg:
[image]

With the terminal comand `fswebcam -r 640x480 --no-banner image.jpg` i get this image:
[image]

Sadly i don't have a external powerd USB-hub or a higher power supply to test the power issue. But the color displacement it everytime exactly the same, which makes me belive it's not a power issue.

I also don't have a second webcam to test this. But as said, with fswebcam it's working, so it can't be the webcam (just maybe a webcam-v4l4j driver incompatibility?).

But even if i tell fswebcam to make several immages and skip the first 1000 frames, the image looks absolutly fine.
 `fswebcam -r 640x480 --no-banner image.jpg -S 1000`
`Captured 1001 frames in 94.86 seconds. (10 fps)`

So even that shouldn't be the problem. Since this command is working fine, i don't belive it's a power or number of pictures issue.

In my eyes it looks more like the red and blue part of the RGB picture arg getting displaced. It they would be placed right, it probably would look fine. Could it be a color space problem? Something that gets converted wrong?

sarxos, 2016-05-31:
Damn. Can you please test this code? I refined webcam-capture-driver-v4l4j code and extracted the most important parts (without format detection which may be to blame). 

It should produce 4 JPG files. Please use the same classpath as you use with webcam-capture-driver-v4l4j.

```java
import com.github.sarxos.v4l4j.V4L4J;

import java.io.File;
import java.io.IOException;
import java.util.concurrent.CountDownLatch;
import java.util.concurrent.TimeUnit;

import javax.imageio.ImageIO;

import au.edu.jcu.v4l4j.CaptureCallback;
import au.edu.jcu.v4l4j.JPEGFrameGrabber;
import au.edu.jcu.v4l4j.VideoDevice;
import au.edu.jcu.v4l4j.VideoFrame;
import au.edu.jcu.v4l4j.exceptions.V4L4JException;

public class Main implements CaptureCallback {

	static {
		V4L4J.init();
	}

	CountDownLatch latch = new CountDownLatch(3);
	int i = 0;

	public static void main(String[] args) throws Exception {
		new Main();
	}

	public Main() throws Exception {

		VideoDevice device = new VideoDevice("/dev/video0");

		JPEGFrameGrabber grabber = device.getJPEGFrameGrabber(640, 480, 0, 0, 80);
		grabber.setCaptureCallback(this);
		grabber.startCapture();

		latch.await(10, TimeUnit.SECONDS);

		grabber.stopCapture();
		device.releaseFrameGrabber();
		device.releaseControlList();
		device.release();
	}

	@Override
	public void nextFrame(VideoFrame frame) {
		try {
			ImageIO.write(frame.getBufferedImage(), "JPG", new File("test-" + (i++) + ".jpg"));
		} catch (IOException e) {
			e.printStackTrace();
		} finally {
			frame.recycle();
			latch.countDown();
		}
	}

	@Override
	public void exceptionReceived(V4L4JException e) {
		e.printStackTrace();
	}
}
```

Spenhouet, 2016-05-31:
Thank you for helping. :+1:

I used the your code but sadly got the same result:
[image]

I will try to get my hands on a USB-Hub with external power supply but probably i will not find one. 

I'm running out of ideas what the problem could be and why it's working with fswebcam but not with the code above, respectively.

I really appreciate your help!

sarxos, 2016-06-01:
Can you please check if the same issue appears on PC? The best would be to test on Linux to have env similar to the RasPi. This can help us verify if this is RasPi specific issue. If this problem appears on PC as well, then ```frame.getBufferedImage()``` may be buggy.

Spenhouet, 2016-06-01:
I don't have a Linux System around and not enough time to set it up in an virtual box.

After you idear of `frame.getBufferedImage()` beeing bugged i replaced the Line

    ImageIO.write(frame.getBufferedImage(), "JPG", new File("test-" + (i++) + ".jpg"));

with this code

    FileOutputStream fos = new FileOutputStream("test-" + (i++) + ".raw");
    try {
        fos.write(frame.getBytes());
    } finally {
        fos.close();
    }

And i got the same bugged image.

I played a little bit more in the small example you provided and changed the FrameGrabber. Seems i found the problem root.

By taking your example code 1:1 and only changing the `JPEGFrameGrabber` to a `RGBFrameGrabber` or `BGRFrameGrabber` solved the issue (Parameter with no Resolution).

[image]

How to change the FrameGrabber for my existing source?

    public File captureFrame() {
        if (!webcam.isOpen())
            webcam.open();

        BufferedImage image = webcam.getImage();
        image = putDateTimeOnImage(image);
        String format = "JPG";
        File file = new File("cam/" + getFileName(format));
        try {
            ImageIO.write(image, format, file);
        } catch (Exception e) {
            Logger.getLogger(PiHardwareController.class.getName()).log(Level.SEVERE, "cam Error", e);
        } finally {
            webcam.close();
        }
        return file;
    }

Spenhouet, 2016-06-01:
Is there a possibility to set the FrameGrabber to `RGBFrameGrabber` or `BGRFrameGrabber` and still get the image with `webcam.getImage()`?
I can't find a way to change the FrameGrabber to the normal setup of this library. If you know a way, help would be appreciated.
Instead i could use the code from above but it's implementation with the event is not what i would need.

Spenhouet, 2016-06-02:
I'm now using it this way:
Cam.java class

    import au.edu.jcu.v4l4j.CaptureCallback;
    import au.edu.jcu.v4l4j.RGBFrameGrabber;
    import au.edu.jcu.v4l4j.VideoDevice;
    import au.edu.jcu.v4l4j.VideoFrame;
    import au.edu.jcu.v4l4j.exceptions.V4L4JException;
    import com.epam.llpd.greenhouse.utils.MyDate;
    import com.github.sarxos.v4l4j.V4L4J;

    import javax.imageio.ImageIO;
    import java.awt.*;
    import java.awt.font.TextLayout;
    import java.awt.image.BufferedImage;
    import java.awt.image.ConvolveOp;
    import java.awt.image.Kernel;
    import java.io.File;
    import java.io.IOException;
    import java.util.concurrent.CountDownLatch;
    import java.util.concurrent.TimeUnit;
    
    /**
     * Created by Spen on 02.06.2016.
     */
    public class Cam implements CaptureCallback {
        static {
            V4L4J.init();
        }

        private CountDownLatch latch = new CountDownLatch(1);
        private String fileName;
        private String fileFormat;
        private File image;

        public Cam(String fileFormat, int width, int height) throws Exception {
            this.fileFormat = fileFormat.toUpperCase();
            VideoDevice device = new VideoDevice("/dev/video0");
            RGBFrameGrabber grabber = device.getRGBFrameGrabber(width, height, 0, 0);
            grabber.setCaptureCallback(this);
            grabber.startCapture();

            latch.await(10, TimeUnit.SECONDS);

            grabber.stopCapture();
            device.releaseFrameGrabber();
            device.releaseControlList();
            device.release();
        }

        @Override
        public void nextFrame(VideoFrame frame) {
            BufferedImage bufferedImage = frame.getBufferedImage();

            BufferedImage image_to_save2 = new BufferedImage(bufferedImage.getWidth(), bufferedImage.getHeight(), BufferedImage.TYPE_INT_RGB);
            image_to_save2.getGraphics().drawImage(bufferedImage, 0, 0, null);
            bufferedImage = image_to_save2;

            image = new File("cam/" + getFileName());
            try {
                ImageIO.write(bufferedImage, fileFormat, image);
            } catch (IOException e) {
                e.printStackTrace();
            } finally {
                frame.recycle();
                latch.countDown();
            }
        }

        @Override
        public void exceptionReceived(V4L4JException e) {
            e.printStackTrace();
        }

        private String getFileName() {
            if (fileName == null || fileName.isEmpty())
                fileName = System.currentTimeMillis() + "." + fileFormat.toLowerCase();

            return fileName;
        }

        public File getImage() {
            return image;
        }
    }

And the useage:

    public File captureFrame() {
        File image = null;
        try {
            image = new Cam("JPG", 640, 480).getImage();
        } catch (Exception e) {
            Logger.getLogger(PiHardwareController.class.getName()).log(Level.SEVERE, "cam Error", e);
        }

        return image;
    }

This is working for me know. I hope it does help in some way, some one.

closed 2016-06-02 by Spenhouet

sarxos, 2016-06-02:
Hi @Spenhouet,

Sorry for late response. Yes, for now there is no other way to replace grabber than relying completely on V4L4J API.

In regards to your code:

```java
BufferedImage image_to_save2 = new BufferedImage(bufferedImage.getWidth(), bufferedImage.getHeight(), BufferedImage.TYPE_INT_RGB);
image_to_save2.getGraphics().drawImage(bufferedImage, 0, 0, null);
bufferedImage = image_to_save2;

image = new File("cam/" + getFileName());
try {
  ImageIO.write(bufferedImage, fileFormat, image);
} catch (IOException e) {
  e.printStackTrace();
} finally {
  frame.recycle();
  latch.countDown();
}
```

I'm wondering why you have to re-draw the image into ```image_to_save2```? I think this should be enough:

```java
BufferedImage bufferedImage = frame.getBufferedImage();
image = new File("cam/" + getFileName());
try {
  ImageIO.write(bufferedImage, fileFormat, image);
} catch (IOException e) {
  e.printStackTrace();
} finally {
  frame.recycle();
  latch.countDown();
  bufferedImage.flush(); // flush reconstructible resources
}
```

However, if the re-draw is necessary then it should be done in the following way. Otherwise you may end up with a memory leak (I think this also depends on Java version) which is pretty hard to diagnose.

```java
BufferedImage image_to_save2 = new BufferedImage(bufferedImage.getWidth(), bufferedImage.getHeight(), BufferedImage.TYPE_INT_RGB);
Graphics2D g2 = image_to_save2.createGraphics();
g2.drawImage(bufferedImage, 0, 0, null);
g2.dispose(); // dispose graphics or you get memory leak

try {
  // do something with image
} finally {
  image_to_save2.flush(); // flush resources or you may end up with a memory leak
}
```

And finally, I see you have file format customized, but in reality there is no sense to save these images as PNG. The only advantage of PNG over JPG is that PNG supports alpha channel where JPG does not. However, you will never get image with alpha channel from webcam, so there is no sense to waste resources to process/save/store PNG files.

Spenhouet, 2016-06-11:
The re-draw wasn't necessary for taking and saving the picture. But i'm printing the date on the picture with Graphics2D and that wasn't working. To re-draw the image like i did, did help with that and Graphics2D did work again. I didn't followed this problem because it got solved by these re-draw lines.

Thanks for the information on the picture format.

Thank you agian for your help!