sarxos/webcam-capture #473: Problem with webcam on odroid

Aurusd, 2016-06-14:
Hi Sarxos. I use your library on Odroid PC board http://www.hardkernel.com/main/products/prdt_info.php?g_code=G143452239825
System: Ubuntu 15.04, java 1.8.91 i586
I have read about The lates version (0.3.10) does not work on ARM just out of the box. To make it working you need to replace version 0.6.2 of BridJ JAR by the 0.6.3-SNAPHOST or newer bridj-0.7-20140918. Moreover, lately Jonathon Hare from OpenIMAJ team, found a problem described in bridj #525 which causes problems on armhf architecture.
Then try sample code with bridj-0.7
public class TakePictureExample {
		public static void main(String[] args) throws IOException {

		// get default webcam and open it
		Webcam webcam = Webcam.getDefault();
	

		/*webcam.setViewSize(WebcamResolution.QVGA.getSize());
		webcam.open();
		// get image
		BufferedImage image = webcam.getImage();
		// save image to PNG file
		ImageIO.write(image, "JPG", new File("test.jpg"));
	}
}

and have next result
Segmentation fault (core dumped).
Maybe you have some advise to me.

sarxos, 2016-06-30:
Hi @Aurusd,

Sorry for late response. For Webcam Capture running on ARM architecture have you tried V4L4J driver? 

https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers/driver-v4l4j

Regards

Aurusd, 2016-06-30:
Hi Sarxos. Thanks for answer, even with late. I have done my issue with bytedeco library. But if you help with your library it would be nice. I added 
```
static {
    Webcam.setDriver(new V4l4jDriver());
}
```
and have next result
```
libvideo.c:68 ] Using libvideo version 0.9-UNKNOWN
[ v4l2-query.c:49 ] Error looking up V4L2 format 34363248. Please submit
a bug report on the v4l4j mailing list.[ v4l2-query.c:484 ] libvideo has skipped an unsupported image format:
[ v4l2-query.c:485 ] H264 Encoded Stream (0x34363248)
[ v4l2-query.c:486 ] Please let the author know about this error.
[ v4l2-query.c:487 ] See the ISSUES section in the libvideo README file.
[ v4l2-query.c:49 ] Error looking up V4L2 format 3447504d. Please submit
a bug report on the v4l4j mailing list.[ v4l2-query.c:484 ] libvideo has skipped an unsupported image format:
[ v4l2-query.c:485 ] MPEG4 Encoded Stream (0x3447504d)
[ v4l2-query.c:486 ] Please let the author know about this error.
[ v4l2-query.c:487 ] See the ISSUES section in the libvideo README file.
[ v4l2-query.c:49 ] Error looking up V4L2 format 33363248. Please submit
a bug report on the v4l4j mailing list.[ v4l2-query.c:484 ] libvideo has skipped an unsupported image format:
[ v4l2-query.c:485 ] H263 Encoded Stream (0x33363248)
[ v4l2-query.c:486 ] Please let the author know about this error.
[ v4l2-query.c:487 ] See the ISSUES section in the libvideo README file.
[ v4l2-query.c:49 ] Error looking up V4L2 format 30385056. Please submit
a bug report on the v4l4j mailing list.[ v4l2-query.c:484 ] libvideo has skipped an unsupported image format:
[ v4l2-query.c:485 ] VP8 Encoded Stream (0x30385056)
[ v4l2-query.c:486 ] Please let the author know about this error.
[ v4l2-query.c:487 ] See the ISSUES section in the libvideo README file.
[ v4l2-query.c:654 ] Error checking supported palettes on V4L2 video device /dev/video7
[ libvideo.c:400 ] libvideo was unable to gather information on device /dev/video7
[ libvideo.c:401 ] If this device is not currently used by any other application,
[ libvideo.c:402 ] please let the author know about this error.
[ libvideo.c:403 ] See the ISSUES section in the libvideo README file.
Exception in thread "main" com.github.sarxos.webcam.Webca
```
In this exception nothing present about my camera /dev/vide0
If don't use any setting of driver have next 
```
SLF4J: Actual binding is of type [ch.qos.logback.classic.util.ContextSelectorStaticBinder]
INFO 2016-06-30 08:10:45,832 com.github.sarxos.webcam.Webcam - WebcamDefaultDriver capture driver will be used
Java HotSpot(TM) Client VM warning: You have loaded library /tmp/BridJExtractedLibraries4565770014616361823/bridj.so which might have disabled stack guard. The VM will try to fix the stack guard now.
It's highly recommended that you fix the library with 'execstack -c <libfile>', or link it with '-z noexecstack'.
java.lang.RuntimeException: Failed to initialize BridJ
 at org.bridj.Platform.initLibrary(Platform.java:370)
 at org.bridj.Platform.<clinit>(Platform.java:175)
 at com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.<clinit>(OpenIMAJGrabber.java:59)
 at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.handle(WebcamDefaultDriver.java:56)
 at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
 at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
 at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
 at java.lang.Thread.run(Thread.java:745)
```
Change version of bridj from 0.6.2 to 0.7.0 code generate no exception and nothing happens

Fomikadze, 2020-10-17:
Hi Sarxos!
I have the same problem with OpenCV driver:

2020-10-17 10:51:55.833 [nioEventLoopGroup-3-2] DEBUG com.github.sarxos.webcam.Webcam - Setting new capture driver JavaCvDriver
OpenCV: out device of bound (0-0): 1
OpenCV: camera failed to properly initialize!
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007fff6ca75678, pid=2257, tid=0x0000000000000307
#
# JRE version: Java(TM) SE Runtime Environment (8.0_181-b13) (build 1.8.0_181-b13)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (25.181-b13 mixed mode bsd-amd64 compressed oops)
# Problematic frame:
# C  [libobjc.A.dylib+0x8678]  objc_release+0x18
#
# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /Users/roman/Projects/video-client/hs_err_pid2257.log
2020-10-17 10:52:03.107 [nioEventLoopGroup-3-2] INFO  c.g.s.webcam.WebcamDiscoveryService - **Discovery will not run - driver JavaCvDriver does not support this feature**
2020-10-17 10:52:03.108 [nioEventLoopGroup-3-2] DEBUG com.github.sarxos.webcam.WebcamLock - Lock Webcam Webcam 0
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
#

Process finished with exit code 134 (interrupted by signal 6: SIGABRT)

My dependencies:
        <dependency>
            <groupId>com.github.sarxos</groupId>
            <artifactId>webcam-capture</artifactId>
            <version>0.3.13-SNAPSHOT</version>
        </dependency>
        <dependency>
            <groupId>com.github.sarxos</groupId>
            <artifactId>webcam-capture-driver-opencv</artifactId>
            <version>0.3.13-SNAPSHOT</version>
        </dependency>
        <dependency>
            <groupId>org.bytedeco</groupId>
            <artifactId>javacpp</artifactId>
            <version>1.5.4</version>
        </dependency>
        <dependency>
            <groupId>org.bytedeco</groupId>
            <artifactId>opencv-platform</artifactId>
            <version>4.4.0-1.5.4</version>
        </dependency>

I'm developing on Mac, so V4L4J drive is not an option.
Maybe there are some other options?