sarxos/webcam-capture #474: Integrate with browser

timpamungkas, 2016-06-20:
Hi sarxos, you have an excellent library here. It works fine on swing-based application.
However, can I use webcam-capture at browser, using ZK frameworks?

I'm trying to access client's webcam, but since webcam codes runs on server side, it always take server's webcam.

Thanks

sarxos, 2016-06-20:
There is no way to use this library in browser without an applet. For webcam integration on web page I recommend https://github.com/jhuckaby/webcamjs

alexmao86, 2016-06-29:
Using java applet in browser is awful in performance. actually, this is very simple in modern browser with HTML5 support.
You just use one IMG tag with src pointing to one JPEG or some other stream. everything is done.

sarxos, 2016-06-29:
Yup, exactly. Applets are already dead and if you think about the future of your product you should make your Java application [web start](http://www.oracle.com/technetwork/java/javase/jws-white-paper-150004.pdf) or use HTML5 instead.

sarxos, 2016-06-30:
Hi guys, I'm closing this ticket since the answer for browser integration question has been already provided.

closed 2016-06-30 by sarxos