sarxos/webcam-capture #475: I have to start webcam in main thread?

celetium, 2016-06-24:
I can't start webcam in an AKKA actor. The error message is: 
> 2016-06-24 18:10:07.611 java[15819:7638897] !!! BUG: The current event queue and the main event queue are not the same. Events will not be handled correctly. This is probably because _TSGetMainThread was called for the first time off the main thread.

I can start webcam successfully in Java main class or Scala App.

sarxos, 2016-06-28:
Hi @celetium,

I cannot confirm. I'm able to use Webcam Capture in Akka without any problem. I guess your issue is caused by something else.

Just wrote this code to create new ```ActorSystem```, spawn ```WebcamActor``` and get ```BufferedImage``` by corresponding ```ActorRef```:

```java
import static akka.pattern.Patterns.ask;
import static scala.concurrent.Await.result;

import java.awt.image.BufferedImage;
import java.io.File;

import javax.imageio.ImageIO;

import scala.concurrent.duration.Duration;
import scala.concurrent.duration.FiniteDuration;
import akka.actor.ActorRef;
import akka.actor.ActorSystem;
import akka.actor.Props;
import akka.actor.UntypedActor;

import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamResolution;
import com.typesafe.config.ConfigFactory;

public class Application {

	static enum GetImageMsg {
		OBJECT;
	}

	static class WebcamActor extends UntypedActor {

		final Webcam webcam;

		public WebcamActor(Webcam webcam) {
			this.webcam = webcam;
		}

		@Override
		public void preStart() throws Exception {
			webcam.setViewSize(WebcamResolution.VGA.getSize());
			webcam.open();
		}

		@Override
		public void postStop() throws Exception {
			webcam.close();
		}

		@Override
		public void onReceive(Object msg) throws Exception {
			if (msg instanceof GetImageMsg) {
				sender().tell(getImage(), self());
			} else {
				unhandled(msg);
			}
		}

		public BufferedImage getImage() {
			return webcam.getImage();
		}
	}

	public static void main(String[] args) throws Exception {

		ActorSystem system = ActorSystem.create("hello-from-akka", ConfigFactory.defaultApplication());
		ActorRef ref = system.actorOf(Props.create(WebcamActor.class, Webcam.getDefault()));

		Duration timeout = FiniteDuration.create("10s");
		BufferedImage image = (BufferedImage) result(ask(ref, GetImageMsg.OBJECT, timeout.toMillis()), timeout);
		ImageIO.write(image, "JPG", new File("test.jpg"));

		system.terminate();
	}
}
```

The effect:

[image]

celetium, 2017-02-08:
Thanks, I will try it.

closed 2017-02-08 by celetium