sarxos/webcam-capture #476: screen freeze when run(using drag and drag components in netbeans)

idontwantosleep, 2016-06-25:
sirr im using java netbeans and i drag and drop the components  for some reason the camera will not open if i put the codes for reading the qr code and if i didnt it will run smoothly, i dont really know what to do, i put this in a method and call it when button is clicked, your my last hope this is my code :

//--------put the camera in jpanel
Webcam webcam = Webcam.getDefault();
webcam.setViewSize(WebcamResolution.VGA.getSize());
WebcamPanel panel = new WebcamPanel(webcam);

       jPanel1.add(panel);
       jPanel1.getParent().revalidate();

       //------READ qr/barcode---------
            do {
        try {
            Thread.sleep(100);
        } catch (InterruptedException e) {
            e.printStackTrace();
        }

        Result result = null;
        BufferedImage image = null;

        if (webcam.isOpen()) {

            if ((image = webcam.getImage()) == null) {
                continue;
            }

            LuminanceSource source = new BufferedImageLuminanceSource(image);
            BinaryBitmap bitmap = new BinaryBitmap(new HybridBinarizer(source));

            try {
                result = new MultiFormatReader().decode(bitmap);
            } catch (NotFoundException e) {
                // fall thru, it means there is no QR code in image
            }
        }

        if (result != null) {
            jTextArea1.setText(result.getText());
        }

    } while (true);

sarxos, 2016-06-28:
I'm not sure if this is Webcam Capture or NetBeans fault. Can you provide more details?

idontwantosleep, 2016-06-30:
sirr i got everything working now, thank you so much your work is a
masterpiece and i really appreciate your response btw i add you on facebook
i hope to see you on my timeline so i could follow your works. -greetings
mark

sarxos, 2016-06-30:
Hi @idontwantosleep, thank you for your feedback. I'm really glad you resolved your problem with the code. Take care! I'm closing this ticket.

closed 2016-06-30 by sarxos