sarxos/webcam-capture #477: How can close ?

RedToor, 2016-06-30:
how to solve ? 
> Cannot change resolution when webcam is open, please close it firstCannot change resolution when webcam is open, please close it first

`
      webcam.close() // no work
`
thanks

sarxos, 2016-06-30:
Hi @RedToor,

What do you mean by ```webcam.close()``` not working? Is there any error, exception, warning in logs?

RedToor, 2016-06-30:
Hi @sarxos for answer, i have this code when I run first time, it is going well, when I call back function does not work

`

			private static void catCam(String string) throws IOException {
		        try{
			    webcam = Webcam.getDefault();
			    Dimension[] nonStandardResolutions = new Dimension[] { 
					WebcamResolution.PAL.getSize(),
					WebcamResolution.HD720.getSize(),
					new Dimension(2000, 1000),
					new Dimension(1000, 500),
				};

			for (Webcam webcam1 : Webcam.getWebcams()) {
				if(webcam1.equals(string)){webcam=webcam1;}}
				webcam.setCustomViewSizes(nonStandardResolutions);
				webcam.setViewSize(WebcamResolution.HD720.getSize());
				webcam.open();                   
				while (InputStream.readUTF().equals("N")){
					WebcamUtils.capture(webcam,TempFolder+"c~tmp~cam", "jpg");
					StreamingImage(TempFolder+"c~tmp~.jpg");
	    		}
				webcam.close();
		} catch (Exception e) {System.out.println(e.getMessage()+e.getLocalizedMessage());} 		
	}
`

Message

> Cannot change resolution when webcam is open, please close it firstCannot change resolution when webcam is open, please close it first

sarxos, 2016-06-30:
Hi @RedToor,

Your code is fundamentally wrong. 

For example this ```if``` - it will never be true:

```
if(webcam1.equals(string)){webcam=webcam1;}}
```

Since you **cannot** compare String and Webcam objects. These are two completely different classes!

Also, doing this for/loop:

```
for (Webcam webcam1 : Webcam.getWebcams()) {
```
Has no effect since, because mentioned ```if``` always gives false, the ```webcam``` will never be assigned.

Do you call this method twice from different threads?

Try this one. Since ```WebcamUtils``` opens webcam if it's not open you don't have to open it by your self. Just remembr to close it - I added one statement for this.

```java
Dimension[] nonStandardResolutions = new Dimension[] { WebcamResolution.HD720.getSize() };

Webcam webcam = Webcam.getDefault();
webcam.setCustomViewSizes(nonStandardResolutions);
webcam.setViewSize(WebcamResolution.HD720.getSize());

WebcamUtils.capture(webcam, "test.jpg", "jpg");

if (webcam.isOpen()) {
	webcam.close();
}
```

RedToor, 2016-06-30:
@sarxos i used `if(webcam1.equals(string)){webcam=webcam1;}}` for set a webcam not default.

sarxos, 2016-06-30:
It will not work. You cannot compare string and webcam. This will always be false. Instead, if you want to get webcam by name just use Webcam.getWebcamByName(name)

RedToor, 2016-06-30:
Yes , is that just what I needed , Thanks.

sarxos, 2016-07-01:
Great :) I will close this ticket. In case you have more questions please ask.

closed 2016-07-01 by sarxos