sarxos/webcam-capture #480: just need help !

PauSeekings, 2016-07-04:
Not an issue,  just be being thick.  
I've looked at the different examples. 
How do I use:	

BufferedImage img = webcam.getImage();

and then display the img  on screen in a window, in a loop? 

I want to average the buffered images over time .. to give a long exposure effect, and do other manipulations, so I need access to  BufferedImage img .
I've looked that examples using the JH filters, thats close.. but there no temporal filters,  i.e. ones that require multiple past frames to calculate the current frame.  

thanks  !

PauSeekings, 2016-07-05:
Took a closer look at the  examples  using JH filters, that does give access to the buffered image in the  	@Override  transform(BufferedImage image) . So I can do my stuff there !  Thanks

closed 2016-07-05 by PauSeekings

sarxos, 2016-07-05:
Great :) Take care!