sarxos/webcam-capture #481: Example using the non-default camera ?

PauSeekings, 2016-07-06:
I've been using this a while on  OS X and raspberry pi.  If I'm on my iMac, the default webcam is the Isight, but I have a second usb webcam that I want to use.  
How to list the available webcams and connect to the USB one. 
Thanks
P

sunnyguan, 2016-07-06:
you can use `Webcam.getWebcams()` to get all the webcams and open the ones you want to use. @sarxos please tell me if I am wrong :)

PauSeekings, 2016-07-07:
`List<Webcam> wb =Webcam.getWebcams();`
Gives me list, of the webcams connected, the first one is default, the second one is the one I want to use. How do specify to open the second in the list ?

sunnyguan, 2016-07-07:
`wb.get(1).open();`

PauSeekings, 2016-07-07:
Thanks  Sunny !!!  just to sum up ....

```
List<Webcam> webcam =Webcam.getWebcams();
// List all webcams connected 
//System.out.println( webcam );
// Default camera,    webcam.get(0)
// Second camera,    webcam.get(1)

webcam.get(1).setViewSize(WebcamResolution.VGA.getSize());
webcam.get(1).open();
```

Will list the multiple webcams connected to the com. and open the second one (in this case);

closed 2016-07-07 by PauSeekings

sarxos, 2016-07-07:
Yes, it's exactly as @sunnyguan stated.

Example:

```java
Webcam second = Webcam.getWebcams().get(1);
second.open();
BufferedImage image = second.getImage();
second.close();
```