sarxos/webcam-capture #482: Problem with professional usb cam fps (not webcam)

jljljl, 2016-07-09:
Hello! I've connected to usb cam using your lib, but here is a problem. Cam's fps is too slow because it sends YUY images, but cam works fast only on MJPEG. How can I do it? How to force MJPEG connection against default? Amcap shows 5x better fps using MJPEG then YUY. Here is some people waiting for your answer.

sarxos, 2016-07-09:
It depends on operating system and capture driver used. What is yours?

jljljl, 2016-07-10:
Win 7x64/default driver (dunno if any driver can use MJPEG)

sarxos, 2016-07-11:
Hi @jljljl, 

On Windows default driver uses OpenIMAJ which is build on top of VideoImput library which does not support MJPEG. Just FYI, I'm implementing MJPEG in capture driver for GStreamer 1.x framework, but it is in development state and only supports Linux by now :(

jljljl, 2016-07-12:
Hi! Can I use RGB24? Or maybe I can help you to develop MJPEG faster? Can I contact you anywhere? I hope you got me

jljljl, 2016-07-17:
Hello @sarxos! This problem needs solution as fast as it's possible. Some ppl are waiting for you and we can help you. Would be great if you'll fill with text ["how to enable debug logs page"](https://github.com/sarxos/webcam-capture/wiki/How-To-Enable-Dubug-Logs). Also, we hope you can allow MJPG connection (or just help RGB24, it might work too).

sarxos, 2016-07-19:
Hi @jljljl, 

If you have spare time you can take ```webcam-capture-driver-gstreamer``` as a base and build driver which uses GStreamer 1.x instead of 0.10. New driver is called ```webcam-capture-driver-gst1``` but it's completely raw righ now (not working).

You have to install GStreamer 1.x (google installer and run it) and build Java bindings:

```plain
$ git clone git@github.com:gstreamer-java/gst1-java-core.git
$ cd gst1-java-core
$ mvn clean install
```

After above is done most of the errors in  ```webcam-capture-driver-gst1``` should disappear.

I cannot work on it right now due my work duties and the fact that I have a wedding in a few days so I will be OOO for about a month. Before I go I will try to put some more code there and make it more usable. Adding Windows support after Linux is ready should be fairy simple if there is no blocking problem.

In regards to enable debug logs page - there was a mistake in link URL (_D**u**bug_ instead of _D**e**bug_). The page exists and it's available under the correct URL:

https://github.com/sarxos/webcam-capture/wiki/How-To-Enable-Debug-Logs

sarxos, 2016-07-19:
https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers/driver-gst1

jljljl, 2016-07-23:
Thx @sarxos, i will try