sarxos/webcam-capture #484: Regarding face recognition and recording video

Jacky412, 2016-07-11:
Hi, first of all thank you for creating such a great API. I am currently working on my final year project and have two burning questions in mind:

1.) I have checked out the example on face recognition i.e. 
https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-examples/webcam-capture-detect-face
However the link does not provide any source code to be downloaded. Could you kindly add in the source code and provide some guidance as to where to download OpenIMAJ? I have visited the official website of OpenIMAJ and cannot not find any jar file to download.

2.) I have also checked out the example on video recording i.e.
https://github.com/sarxos/webcam-capture/blob/master/webcam-capture-examples/webcam-capture-video-recording/src/main/java/com/github/sarxos/webcam/Encoder.java
It seen like you have imported library from xuggle-xuggler which is now deprecated. May I know is it possible to record video without importing this library? If not where could I download the xuggle-xuggler jar file?

Thanks in advance. Greatly appreciate if you could help.

Best Regards,
Jacky

sarxos, 2016-07-12:
Hi @Jacky412,

In regards to your comments:

> However the link does not provide any source code to be downloaded

There is no link to download source because there is no need for it. If you need source just clone the project from Github or download it as ZIP. Then, after you have it extracted on your desktop, just go to the appropriate directory where sources are located:

```plain
$ cd webcam-capture
$ cd webcam-capture-examples/webcam-capture-detect-face 
$ cd src/main/java/com/github/sarxos/webcam
$ ls -la
total 12
drwxrwxr-x 2 sarxos sarxos 4096 wrz 18  2014 .
drwxrwxr-x 3 sarxos sarxos 4096 wrz 18  2014 ..
-rw-rw-r-- 1 sarxos sarxos 3307 wrz 18  2014 FacePainterExample.java
```

Here you are, the ```FacePainterExample.java``` which contains the source code (but not whole) for this example.

The source code is also available via Github:

[image]

And in regards to this:

>  May I know is it possible to record video without importing this library?

Yes, but you have to use another library for recording. Webcam Capture does not provide any recording functionality ad hoc.

> If not where could I download the xuggle-xuggler jar file?

I see that repository URL from [```pom.xml```](https://github.com/sarxos/webcam-capture/blob/master/webcam-capture-examples/webcam-capture-video-recording/pom.xml) is not active any more, so I'm sorry to disappoint you - I don't know where you can download it nowadays :( With this Xuggler repository broken I will need to think about removing this example or changing it so it uses some different tooling for recording.

Jacky412, 2016-07-12:
Hi Sarxos,

Thanks a lot for your prompt and detailed reply. I managed to download the source code for the example on face recognition (i.e. "FacePainterExample.java") however the following exception was thrown during compilation with Eclipse IDE on Windows 10:

Exception in thread "main" java.lang.NoClassDefFoundError: org/openimaj/io/ReadWriteableBinary
	at java.lang.ClassLoader.defineClass1(Native Method)
	at java.lang.ClassLoader.defineClass(Unknown Source)
	at java.security.SecureClassLoader.defineClass(Unknown Source)
	at java.net.URLClassLoader.defineClass(Unknown Source)
	at java.net.URLClassLoader.access$100(Unknown Source)
	at java.net.URLClassLoader$1.run(Unknown Source)
	at java.net.URLClassLoader$1.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(Unknown Source)
	at java.lang.ClassLoader.loadClass(Unknown Source)
	at sun.misc.Launcher$AppClassLoader.loadClass(Unknown Source)
	at java.lang.ClassLoader.loadClass(Unknown Source)
	at java.lang.ClassLoader.defineClass1(Native Method)
	at java.lang.ClassLoader.defineClass(Unknown Source)
	at java.security.SecureClassLoader.defineClass(Unknown Source)
	at java.net.URLClassLoader.defineClass(Unknown Source)
	at java.net.URLClassLoader.access$100(Unknown Source)
	at java.net.URLClassLoader$1.run(Unknown Source)
	at java.net.URLClassLoader$1.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(Unknown Source)
	at java.lang.ClassLoader.loadClass(Unknown Source)
	at sun.misc.Launcher$AppClassLoader.loadClass(Unknown Source)
	at java.lang.ClassLoader.loadClass(Unknown Source)
	at WebCam.FacePainterExample.<clinit>(FacePainterExample.java:33)
Caused by: java.lang.ClassNotFoundException: org.openimaj.io.ReadWriteableBinary
	at java.net.URLClassLoader.findClass(Unknown Source)
	at java.lang.ClassLoader.loadClass(Unknown Source)
	at sun.misc.Launcher$AppClassLoader.loadClass(Unknown Source)
	at java.lang.ClassLoader.loadClass(Unknown Source)
	... 25 more

I am using your webcam capture API as well as OpenIMAJ libraries downloaded from:

http://maven.ecs.soton.ac.uk/content/repositories/openimaj-releases/org/openimaj/faces/1.3.5/
http://maven.ecs.soton.ac.uk/content/repositories/openimaj-releases/org/openimaj/core-image/1.3.5/
http://maven.ecs.soton.ac.uk/content/repositories/openimaj-releases/org/openimaj/core-math/1.3.5/

Please kindly advise. Thank you for your time and patience.

Best Regards,
Jacky

sarxos, 2016-07-12:
You probably did this manually? Am I right? This happens if you want to manage transitive dependencies on your own instead of using Maven/Gradle or Ivy.

If you want to import all libraries manually (but I **really** suggest to use Maven), here is the list:

http://pastebin.com/0QBaG2eq

You can go to the Maven Central and download these JARs on-by-one (which is a pain in the ass):

http://search.maven.org/

Or you can use Maven to download all of them to a directory:

```
mvn dependency:copy-dependencies -DoutputDirectory=libs
```

Or finally, if you are too lazy to install Maven and do the above, just download this ZIP file which contains the files I downloaded when I ran the above command:

https://drive.google.com/open?id=0ByUTQ5eXduvFMWRSUS1XVWtObDA

Jacky412, 2016-07-13:
Hi Sarxos,

Thanks again for your reply. To be honest I have never used Maven before and have zero knowledge on how to install and setup which is why I choose to download each jar file manually. Nevertheless after much researching and googling I finally managed to install Maven on Eclipse IDE and manage to compile the code without any error.

However there is one last problem, the file size. I have tried exporting the project into executable jar and the size of the jar is 117 MB for just the example "FacePainterExample.java".

I am currently working on my final year project and will be really grateful if you could help me out with this one last problem.

Best Regards,
Jacky

sarxos, 2016-07-13:
Hi @Jacky412,

That's good you managed to start using Maven. I helps you manage dependencies and makes your developer life easier :)

In regards to the size. Yes, OpenIMAJ feature detection has a lot of transitive dependencies (JARs) which are not required (e.g. stuff for azure and other non related functions), but you will have to test manually which ones are these by excluding them one-by-one or several at once and checking if you application works. I did it some time ago and managed to go down to several megabytes only.

Jacky412, 2016-07-14:
Hi Sarxos,

I have went through the jar files one by one and manage to get the minimum set of jar files required to get the program running. The file size now is only around 9 MB.

For the benefit of the others who face the same problem as me the list of jar files are as follows:

core-1.3.5.jar
core-image-1.3.5.jar
core-math-1.3.5.jar
faces-1.3.5.jar
image-processing-1.3.5.jar
log4j-1.2.17.jar
object-detection-1.3.5.jar
sanselan-0.97-incubator.jar
xmlpull-1.1.3.1.jar
xpp3_min-1.1.4c.jar

Best Regards,
Jacky

smarkm, 2016-08-06:
the maven repo `http://xuggle.googlecode.com/svn/trunk/repo/share/java/`  for xuggle-xuggler  invalid

sandeepparek, 2016-09-22:
Well It is a thanks giving project.
it worked.
and
i love that.