sarxos/webcam-capture #485: Segmentation fault on Raspberry PI

LTNico, 2016-07-15:
Hi, I'm sorry but I don't speak English verry well...

I have a segmentation fault when I run my program on my Raspberry PI (3) but it's works on Windows.
I use "webcam-capture 0.3.10" and "bridj 0.7".

`
public class CameraStream extends Thread {
	private final Camera camera;
	
	public CameraStream(Camera camera) {
		this.camera = camera;
		this.start();
	}
	
	public void run() {
		Webcam camera = Webcam.getDefault();
		if (camera == null) {
			System.err.println("Pas de caméra.");
			return;
		} else {
			System.out.println("Caméra : " + camera.getName());
		}
		camera.setViewSize(new Dimension(320, 240));
		camera.open();
	}
}
`

Does anyone know of any solution to this problem ?

LTNico, 2016-07-16:
Solved : Driver webcam-capture-driver-v4l4j

timmattison, 2020-08-04:
For anyone curious you'll need to do this:

```
Webcam.setDriver(V4l4jDriver.class);
```

And you'll also need to add the https://mvnrepository.com/artifact/com.github.sarxos/webcam-capture-driver-v4l4j/0.3.12 library.

timmattison, 2020-08-04:
However, it looks like on the Raspberry Pi 4 when it tries to discover all of the cameras it runs into issues when it is querying the private controls and reports "we shouldnt be here...". I'm working on fixing that.

Longoon12000, 2020-10-30:
Is there any workaround for the "we shouldnt be here..." thing? I can't get it to work even with using `Webcam.getWebcamByName("/dev/video0")`.

timmattison, 2020-10-30:
@Longoon12000 can you post a permalink to the line of code that's throwing this exception/error? I can't find it.

Longoon12000, 2020-10-30:
Oh you're right, it's not even an issue in the webcam-capture repo, it's an issue in the v4l4j repo: https://github.com/mailmindlin/v4l4j/blob/beta/libvideo/v4l2-input.c#L1438
I'll have to bump this issue then: https://github.com/mailmindlin/v4l4j/issues/26

Zekix27, 2021-06-19:
Someone found a solution for "We souldnt be here ..." ?

cjritola, 2025-04-25:
Perhaps "duplicate" would be a better tag than "SOLVED" for this issue as the v4l4j issue being reference has not resolved.