sarxos/webcam-capture #492: License Type

Jacky412, 2016-08-11:
Hi Sarxos,

I am aware that this software has been declared free of charge without limitation to use. However since I am using your API in my final year project I'll need to document it properly as to what type of license it belongs to. So it is considered as freeware, GPL or LGPL...?

Thanks again for creating such a wonderful piece of API. You are truly a genius.

Best Regards,
Jacky

wodin, 2016-11-27:
According to the LICENSE.txt it is an MIT license:

https://github.com/sarxos/webcam-capture/blob/master/LICENSE.txt

sarxos, 2016-11-29:
Hi @Jacky412,

It's a [MIT](https://en.wikipedia.org/wiki/MIT_License) license, as @wodin stated. The only exception from MIT is [webcam-capture-driver-v4l4j](https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers/driver-v4l4j) which is [GNU GPL v3](https://en.wikipedia.org/wiki/GNU_General_Public_License#Version_3) derived from a [V4L4J](https://code.google.com/archive/p/v4l4j/) project used in this specific driver classpath.

Good luck with your final year project! :+1:

closed 2016-11-29 by sarxos