sarxos/webcam-capture #495: Raspberry Pi 3 - getImage is null - plenty of info attached

limpygnome, 2016-08-21:
webcam.getImage() is returning null after some time, looks to be timing out.

This is using a Raspberry Pi 3 with Debian Jesse (minimum), with libjpeg8-dev and libv4l-dev dependencies installed. Webcam is a Logitech C170, which is definitely working, since e.g. "fswebcam test.jpg" is using /dev/video0. 

The program works fine on Ubuntu 14.04.

Current version:
> pi@raspberrypi:~ $ uname -r
> 4.4.13-v7+

> pi@raspberrypi:~ $ java -version
> openjdk version "1.8.0_40-internal"
> OpenJDK Runtime Environment (build 1.8.0_40-internal-b04)
> OpenJDK Zero VM (build 25.40-b08, interpreted mode)

POM dependencies:
```
<!-- Webcam API -->
        <dependency>
            <groupId>com.github.sarxos</groupId>
            <artifactId>webcam-capture</artifactId>
            <version>0.3.11</version>
        </dependency>

        <!-- Support for Pi -->
        <dependency>
            <groupId>com.nativelibs4java</groupId>
            <artifactId>bridj</artifactId>
            <version>0.7.0</version>
        </dependency>
        <dependency>
            <groupId>com.github.sarxos</groupId>
            <artifactId>webcam-capture-driver-v4l4j</artifactId>
            <version>0.3.11</version>
        </dependency>
        <!-- https://github.com/sarxos/webcam-capture/wiki/How-To-Configure-Raspberry-Pi -->
        <dependency>
            <groupId>com.github.sarxos</groupId>
            <artifactId>v4l4j</artifactId>
            <version>0.9.1-r507</version>
        </dependency>

        <!-- webcam-capture log libs -->
        <dependency>
            <groupId>ch.qos.logback</groupId>
            <artifactId>logback-core</artifactId>
            <version>1.0.9</version>
        </dependency>
        <dependency>
            <groupId>ch.qos.logback</groupId>
            <artifactId>logback-classic</artifactId>
            <version>1.0.9</version>
        </dependency>
```

Code:
```
static
    {
        System.setProperty("org.slf4j.simpleLogger.defaultLogLevel", "debug");
        System.setProperty("org.slf4j.simpleLogger.log.com.github.sarxos.webcam.ds.v4l4j", "trace");

        WebcamLogConfigurator.configure("logback.xml");

        Webcam.setDriver(new V4l4jDriver());
    }

    public static void main(String[] args) throws Exception
    {
        System.out.println("starting execution...");
[... 103 lines cut ...]
> 21:46:59.055 [atomic-processor-1] ERROR c.g.s.webcam.ds.v4l4j.V4l4jDevice - UNable to get image in {} seconds timeout
> 21:47:00.272 [main] ERROR c.github.sarxos.webcam.WebcamUpdater - Image has not been found for more than 10 seconds
> attempt #1...
> 21:47:02.058 [atomic-processor-1] ERROR c.g.s.webcam.ds.v4l4j.V4l4jDevice - UNable to get image in {} seconds timeout
> 21:47:05.061 [atomic-processor-1] ERROR c.g.s.webcam.ds.v4l4j.V4l4jDevice - UNable to get image in {} seconds timeout
> 21:47:08.064 [atomic-processor-1] ERROR c.g.s.webcam.ds.v4l4j.V4l4jDevice - UNable to get image in {} seconds timeout
> 21:47:10.485 [main] ERROR c.github.sarxos.webcam.WebcamUpdater - Image has not been found for more than 10 seconds
> attempt #2...
> 21:47:11.066 [atomic-processor-1] ERROR c.g.s.webcam.ds.v4l4j.V4l4jDevice - UNable to get image in {} seconds timeout
> 21:47:14.069 [atomic-processor-1] ERROR c.g.s.webcam.ds.v4l4j.V4l4jDevice - UNable to get image in {} seconds timeout
> 21:47:17.072 [atomic-processor-1] ERROR c.g.s.webcam.ds.v4l4j.V4l4jDevice - UNable to get image in {} seconds timeout
> 21:47:20.075 [atomic-processor-1] ERROR c.g.s.webcam.ds.v4l4j.V4l4jDevice - UNable to get image in {} seconds timeout
> 21:47:20.697 [main] ERROR c.github.sarxos.webcam.WebcamUpdater - Image has not been found for more than 10 seconds
> attempt #3...
> 21:47:23.078 [atomic-processor-1] ERROR c.g.s.webcam.ds.v4l4j.V4l4jDevice - UNable to get image in {} seconds timeout
> 21:47:26.080 [atomic-processor-1] ERROR c.g.s.webcam.ds.v4l4j.V4l4jDevice - UNable to get image in {} seconds timeout
> 21:47:29.083 [atomic-processor-1] ERROR c.g.s.webcam.ds.v4l4j.V4l4jDevice - UNable to get image in {} seconds timeout
> 21:47:30.909 [main] ERROR c.github.sarxos.webcam.WebcamUpdater - Image has not been found for more than 10 seconds
> attempt #4...
> 21:47:32.086 [atomic-processor-1] ERROR c.g.s.webcam.ds.v4l4j.V4l4jDevice - UNable to get image in {} seconds timeout
> 21:47:35.089 [atomic-processor-1] ERROR c.g.s.webcam.ds.v4l4j.V4l4jDevice - UNable to get image in {} seconds timeout
> 21:47:38.092 [atomic-processor-1] ERROR c.g.s.webcam.ds.v4l4j.V4l4jDevice - UNable to get image in {} seconds timeout
> 21:47:41.095 [atomic-processor-1] ERROR c.g.s.webcam.ds.v4l4j.V4l4jDevice - UNable to get image in {} seconds timeout
> 21:47:41.122 [main] ERROR c.github.sarxos.webcam.WebcamUpdater - Image has not been found for more than 10 seconds
>

limpygnome, 2016-08-21:
Seems to be an issue with OpenJDK, since this works with oracle-jdk-8.

kemal409, 2016-08-22:
I was experiencing a similar issue where it was working on my desktop but couldn't get it to work with the raspberry pi. After removing openjdk8 and installing oracle-java8-jdk problem is solved. Maybe this should be stated on how to configure raspberry pi wiki.