sarxos/webcam-capture #498: JVM crash after open/close/open

jlbrandt23, 2016-09-20:
I'm testing on Windows 10 on a Surface Pro 4, with built-in camera.

Here is my test case:
`package client;

import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamPanel;

public class ErrorCase
{        
         private Webcam webcam;
         private WebcamPanel webcamPanel;
         
        public ErrorCase() {      
        webcam = Webcam.getDefault();

       System.out.println("trying to open");
      webcam.open(); 
      System.out.println("opened");
        System.out.println("trying to close");
      webcam.close();
        System.out.println("closed");
        System.out.println("getting camera");
        webcam = Webcam.getDefault();
        System.out.println("trying to open");
      webcam.open();
        System.out.println("opened again");
        System.out.println("trying to close");
      webcam.close();       
        System.out.println("closed again");
        System.exit(0);
      }        
                
      
      public static void main(String[] args)
          {                        
           ErrorCase e = new ErrorCase();
          }
}
`

Here is the error:
Exception in thread "main" com.github.sarxos.webcam.WebcamException: Cannot execute task
        at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.process(WebcamProcessor.java:72)
        at com.github.sarxos.webcam.WebcamProcessor.process(WebcamProcessor.java:140)
        at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:46)
        at com.github.sarxos.webcam.ds.cgt.WebcamOpenTask.open(WebcamOpenTask.java:20)
        at com.github.sarxos.webcam.Webcam.open(Webcam.java:302)
        at com.github.sarxos.webcam.Webcam.open(Webcam.java:251)
        at com.github.sarxos.webcam.Webcam.open(Webcam.java:236)
        at client.ErrorCase.<init>(ErrorCase.java:35)
        at client.ErrorCase.main(ErrorCase.java:46)
Caused by: com.github.sarxos.webcam.WebcamException: Cannot start native grabber!
        at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice.open(WebcamDefaultDevice.java:334)
        at com.github.sarxos.webcam.ds.cgt.WebcamOpenTask.handle(WebcamOpenTask.java:38)
        at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
        at java.lang.Thread.run(Unknown Source)

My use cases are....A) Use may use webcam only periodically.  I'd like to close webcam in between uses.  2) I want to allow uses to change resolution, and it is my understanding webcam must be closed to change resolution.

Is there a preferred way to close & re-open the webcam?

jlbrandt23, 2016-09-20:
I see a possible similar issue in #421.

ChenzhengUESTC, 2016-09-29:
I've got exactly the very same  problem today.

jlbrandt23, 2016-12-12:
I ran the same test case through, but used the NativeDriver from https://github.com/frankpapenmeier/webcam-capture-driver-native and it looks like it is working.   @ChenzhengUESTC , I would recommend trying the Native driver if you are developing for Windows 8 or 10.

closed 2016-12-12 by jlbrandt23