sarxos/webcam-capture #506: TakeSnaphot

Dejiola, 2016-10-23:


Dejiola, 2016-10-24:
Hi Sarxos,
First I must commend you for a job well done. I have been using your webcam programs in my projects and I found their performance very satisfactory. I want to ask is there any way I can use setDefaultCloseOperation(JFrame.DISPOSE_ON_CLOSE) in  TakeSnaphotFromVideoExample program and deactivate the webcam as soon as the window is closed?

Crigges, 2016-10-27:
If you are refering to https://github.com/Fersca/Myron/blob/master/src/TakeSnaphotFromVideoExample.java
You need to shutdown the executor service first and close the webcam afterwards. So you can't use "JFrame.DISPOSE_ON_CLOSE" anymore and need to define your own close operation, see:
http://stackoverflow.com/questions/6084039/create-custom-operation-for-setdefaultcloseoperation

sarxos, 2016-11-29:
Hi @Dejiola,

You can do that by adding window listener into your ```JFrame``` and issue ```webcam.close()``` in ```windowClosed()``` method:

Example:

```java
public class SomeWindow extends JFrame implements WindowListener {

	Webcam webcam = Webcam.getDefault();

	public SomeWindow() {
		addWindowListener(this);
	}

	@Override
	public void windowClosed(WindowEvent e) {
		webcam.close();
	}

	@Override
	public void windowOpened(WindowEvent e) {
		// nothing
	}

	@Override
	public void windowClosing(WindowEvent e) {
		// nothing
	}

	@Override
	public void windowIconified(WindowEvent e) {
		// nothing
	}

	@Override
	public void windowDeiconified(WindowEvent e) {
		// nothing
	}

	@Override
	public void windowActivated(WindowEvent e) {
		// nothing
	}

	@Override
	public void windowDeactivated(WindowEvent e) {
		// nothing
	}
}
```

In case of additional questions please do not ask them here :)

closed 2016-11-29 by sarxos