sarxos/webcam-capture #508: Unable to stream video over public Network

joginderMalik, 2016-11-06:
Hello 
I used https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-examples/webcam-capture-live-streaming this link and its working properly when we do streaming on a private Network.
Issue: i want to do live streaming from few webcams to a single machine. All machine are on public network and there is a router in between. To do this If i have a static public IP from ISP end and i have to forward port 20000 manually.

But if there is no static ip then this code doent work. I spent around 10 days on this but no luck yet. Finally expecting a smart idea from all of you.
Thanks in advance
joginder.malik20@gmail.com

sarxos, 2016-11-29:
The only option I see is to stream from camera to some sink server in public network where stream is re-encoded and made public. However there is no such possibility with the existing implementation. Hopefully someone can implement this mechanism and share it to the community.

joginderMalik, 2016-12-01:
Hey
Thanks for your reply over. I have found a solution to stream over public
network too, I used socket programming to achieve it.

Simply start a ServerSocket then we write buffered image into output stream
and get buffered images into input stream on client side.

It is working now.

Thanks
regards
Joginder malik insan

rakkatak, 2017-09-12:
@joginderMalik would it be possible to see some code samples? thank you! anita

joginderMalik, 2017-09-18:
Hi, i do have a lot of code for different functionalities. We achieved the same in web app and standalone app as well. Going forward please mail me up your requirement that will help me in sharing the relevant/Exact code.
attaching a zip file as of now
[Streaming.zip](https://github.com/sarxos/webcam-capture/files/1311017/Streaming.zip)

email :- joginder.malik20@gmail.com