sarxos/webcam-capture #509: Erro com o driver V4l4j on raspberry

ronaldoSf, 2016-11-16:
Sorry for my english.
I'm triyng to use V4l4j driver on raspberry.
I cant get it to work. It shows me an error : Unable to open device file /dev/video0.
It works with fswebcam driver but it has a very low fps.

Then i've ran the jar with sudo and it doesn't work
Then i've ran chmod 777 /dev/video0 and it doesn't work
Then i decided to get deeper into the error and the problem is in libvideo/libvideo.c in this part:

>`       //open device
	dprint(LIBVIDEO_SOURCE_VIDDEV, LIBVIDEO_LOG_DEBUG,
			"VD: Opening device file %s.\n", file);
	if ((strlen(file) == 0) || ((fd = open(file,O_RDWR )) < 0)) {
		info("V4L: unable to open device file %s. Check the name and "
				"permissions\n", file);
		return NULL;
	}
`
>
It just dont open the file, even with permissions, can someone give some help?

Here the print screen:
[image]

ronaldoSf, 2016-11-17:
I've solved this and im leaving the answer here if someone also has the same case.
The problem is that my system was creating two files /dev/vide0 and /dev/video1. Apparently fswebcam driver chooses the right one: video0. The v4l4j choses the wrong one: video1.
So i deleted video1 so that the v4k4j can choose the right file.
I believe that this is a problem of the system that create two files wrongly. And this only works when i restart the system with  the webcam on the usb, in other words, plug and play is not working

trbLeeciN, 2017-03-09:
ronaldo how is fps? is it slow when using v4l4j? I am going to change my project but need to be sure if it is ok or not

ronaldoSf, 2017-03-09:
No, its not slow. V4l4j has normal fps.