sarxos/webcam-capture #511: Would love support with D-Link DCS933L with webcam-capture-driver-ipcam

85frankenstein, 2016-11-26:
I've been trying for some time to get a program that uses Webcam-Capture working with the D-Link 933L IP Camera with no luck... The developer referred me back to here for help... I would LOVE to get a class working with this particular IP Camera... here in the States, it's cheap, commonly found, and easy to work with...

I do not have access to a live feed or an API, but it works pretty much like many others out there...

What I can provide is the URL that allows you to view the video feed without any of the D-Link Wrappers...

http://[USERNAME]:[PASSWORD]@IPADDRESS:PORT/mjpeg.cgi

85frankenstein, 2016-12-01:
Ok... Since I didn't get any responses, how would one go about building a class for the camera? I have a REAL need to get this running with my IP Camera

sarxos, 2016-12-03:
Hi @85frankenstein, 

If you have MJPEG video feed URL, why not simply use ```IpCamDeviceRegistry.register(name, url, mode, auth)``` as in this example:

https://github.com/sarxos/webcam-capture/blob/master/webcam-capture-drivers/driver-ipcam/src/examples/java/FoscamMjpegPushExample.java

85frankenstein, 2016-12-04:
@sarxos I'm very new to Java... I'm pretty resourceful and very good at Networking but have never had the need to edit code in Java, so please bear with me... I'll take a whack at it and we can see what happens...

sarxos, 2016-12-04:
Hi @85frankenstein, do you have some demo URL, some kind of freely available endpoint or public IP where I can play with this camera?

sarxos, 2016-12-04:
I wanted to try demo from D-Link but I can neither access it from by Ubuntu (they require Internet Explorer, shame on them) nor from my Windows XP virtual machine with IE 8 (probably it's too old and their page doesn't work due to JavaScript errors).

85frankenstein, 2016-12-04:
I'll dig around and see if I can come up with one... Mine isn't public, and I won't port forward, but I'm sure, somewhere, there is an example

85frankenstein, 2016-12-04:
Yeah, the demo cam at D-Link is down for whatever reason... Can't seem to find one in the wild... Might have to put mine up in the DMZ for a very short while... How long would you need @sarxos ?

sarxos, 2016-12-04:
@85frankenstein, few hours should be ok. I need to check how login work in this specific camera (is it base auth or something else), what is the MJPEG stream URI, and finally check if it play nice with [webcam-capture-driver-ipcam](https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers/driver-ipcam). That should be all.

Today is pretty late for me so if possible could we try to find these details tomorrow or after tomorrow?

We can chat on Google Hangouts or you can just drop me an email (bartoszfiryn at gmail dot com) or send Facebook message, so you won't have to put your IP and camera's login credentials into public.

85frankenstein, 2016-12-06:
@sarxos I sent over the link to the video feed without any HTML Wrappers via Google Hangouts... Let me know if you have any troubles

sarxos, 2016-12-06:
Hi @85frankenstein,

With 64fdd35d9f44f2866997771db6b380771c510a5c you can use ```DCS933L``` class to abstract your camera and add it into the IP cameras registry of IP camera capture driver.

```java
IpCamDeviceRegistry.register(new DSC933L(name, url, user, pass));
```

Example:

```java
import java.net.MalformedURLException;
import javax.swing.JFrame;
import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamPanel;
import com.github.sarxos.webcam.ds.ipcam.IpCamDeviceRegistry;
import com.github.sarxos.webcam.ds.ipcam.IpCamDriver;
import com.github.sarxos.webcam.ds.ipcam.device.dlink.DSC933L;

public class DLinkDsc933LExample {

	static {
		Webcam.setDriver(new IpCamDriver());
	}

	public static void main(String[] args) throws MalformedURLException {

		final String name = "D-Link 993L Camera";
		final String user = "{username}"; // change to your own username
		final String pass = "{password}"; // change to your own password
		final String url = "http://{ip-address-or-domain-name}"; // camera's IP address or domain

		IpCamDeviceRegistry.register(new DSC933L(name, url, user, pass));

		final Webcam webcam = Webcam.getDefault();
		final WebcamPanel panel = new WebcamPanel(webcam);

		JFrame f = new JFrame(name);
		f.add(panel);
		f.pack();
		f.setVisible(true);
		f.setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);
	}
}
```

If you do not use programmatic approach but configure your IP cameras in XML registry storage, then it has to be defined as:

```xml
<?xml version="1.0" encoding="UTF-8" ?> 
<storage>
  <ipcam name="{camera-name}" url="http://{ip-address-or-domain-name}/video/mjpg.cgi" mode="push">
    <auth user="{username}" password="{password}" />
  </ipcam>
  <!-- other ip cameras has to be listed here -->
</storage>
```

_(parameters in curly braces has to be replaced by real values)_

sarxos, 2016-12-06:
Newest JAR is already available from Sonatype snapshots repository:

https://oss.sonatype.org/content/repositories/snapshots/com/github/sarxos/webcam-capture-driver-ipcam/0.3.12-SNAPSHOT/webcam-capture-driver-ipcam-0.3.12-20161206.184813-3.jar

Or as a Maven dependency:

```xml
<dependency>
  <groupId>com.github.sarxos</groupId>
  <artifactId>webcam-capture-driver-ipcam</artifactId>
  <version>0.3.12-SNAPSHOT</version>
</dependency>
```

Note that SNAPSHOT version is available only with snaphots repository included in ```pom.xml```:

```xml
<repositories>
  <repository>
    <id>snapshots-repo</id>
    <url>https://oss.sonatype.org/content/repositories/snapshots</url>
    <releases>
      <enabled>false</enabled>
    </releases>
    <snapshots>
      <enabled>true</enabled>
    </snapshots>
  </repository>
</repositories>
```

If you do not use Maven, then simply download this ZIP assembly:

[webcam-capture-driver-ipcam-0.3.12-SNAPSHOT-dist.zip](https://github.com/sarxos/webcam-capture/files/634769/webcam-capture-driver-ipcam-0.3.12-SNAPSHOT-dist.zip)

Required JARs can be found in this zip:

```plain
webcam-capture-driver-ipcam-0.3.12-SNAPSHOT.jar
```

With dependencies:

```plain
libs/bridj-0.7.0.jar
libs/commons-codec-1.6.jar
libs/commons-logging-1.1.1.jar
libs/httpclient-4.2.3.jar
libs/httpcore-4.2.2.jar
libs/httpmime-4.2.3.jar
libs/slf4j-api-1.7.2.jar
libs/webcam-capture-0.3.12-SNAPSHOT.jar
```

If you have any additional questions fell free to ask them here. You can move your camera from DMZ. I won't use it any more.

[image]

Take care!

85frankenstein, 2016-12-06:
Dude! You're awesome!!!!

closed 2016-12-08 by sarxos