sarxos/webcam-capture #514: Does not detect all available ViewSizes for cam

snimavat, 2016-11-30:
I am doing webcam.getViewSizes() to get all supported view sizes for my built in Macbook pro camera.
The highest dimension it returns is 640 * 480. 

But when i use other camera softwares, i see that there are many higher view sizes available for my camera -- (upto 1920 * 1080)

Why does saraxos camera capture doesnt detect all supported view sizes for the camera.

sarxos, 2016-12-03:
Hi @snimavat, in regards to your question:

> Why does saraxos camera capture doesnt detect all supported view sizes for the camera.

Some drivers can detect possible resolutions, e.g. [webcam-capture-driver-v4l4j](https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers/driver-v4l4j), but some cannot sue to limitation of the framework used. If you use default driver which uses OpenIMAJ under the hood, then you are limited to QQVGA, QVGA and VGA because OpenIMAJ does not support resolution detection (at least the version the driver is build on).

If you know that your camera support given resolution, you can simply call:

```
Webcam w = Webcam.getDefault();
w.setCustomViewSizes(new Dimension[] { new Dimension(1920, 1080) });
```

snimavat, 2016-12-03:
@sarxos would that driver work with laptop integrated cameras and usb webcams ?

sarxos, 2016-12-03:
On Apple? Probably not. This specific driver (webcam-capture-driver-v4l4j) uses V4L which is available on Linux platforms, but probably not on Apples. On various Linux distributions this driver works pretty well with both integrated and external UVC devices.

You may try [webcam-capture-driver-jmf](https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers/driver-jmf) with JMF from Oracle, but I cannot guarantee it working on Apple machine because I have no access to MacOS where I can test it running.

snimavat, 2016-12-05:
Thanks for reply @sarxos -- which driver would work on windows, and detect resolutions.. ?

sarxos, 2016-12-05:
Hi @snimavat, [webcam-capture-driver-native](https://github.com/frankpapenmeier/webcam-capture-driver-native) should. It's a very good piece of code. @frankpapenmeier designed it to run on different Windows versions. Instructions how to use it can be found in a project link.

Also [webcam-capture-driver-gstreamer](https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers/driver-gstreamer) and [webcam-capture-driver-jmf](https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers/driver-jmf) are able to detect resolutions on Windows (and Linux as well), but in case of first one, the GStreamer 0.10 (not 1.0) has to be installed on a PC, and for second one the JMF from Oracle has to be installed.

frankpapenmeier, 2016-12-05:
Hi @snimavat, sarxos is right, please have a look at [webcam-capture-driver-native](https://github.com/frankpapenmeier/webcam-capture-driver-native). It works with both Windows and Mac. If you want to quickly evaluate whether it works with your webcams as expected, I suggest that you download the baby monitor app "BabyPhone Mobile" for Windows/Mac: http://www.babyphonemobile.com/eng/download because it uses it under the hood. Just plug in your webcam, start BabyPhone Mobile, hit the settings icon on the right and ensure that "HD capture" is selected for the "Choose webcam capture mode" preference. Then start it as "Sender" and you will see the captured camera stream. If this works, then using [webcam-capture-driver-native](https://github.com/frankpapenmeier/webcam-capture-driver-native) for your project should be well worth a try.

eduramiba, 2021-06-17:
Hi,

Please try the new native driver that I just released: https://github.com/eduramiba/webcam-capture-driver-native

It should be able to correctly list all device resolutions.