sarxos/webcam-capture #521: Error while Webcam.getDefault();

MatisiekPL, 2016-12-30:
Hello, I have a problem with my java program:
On start I have this StackTrace:

Exception in thread "main" java.lang.NoClassDefFoundError: com/github/sarxos/webcam/Webcam
        at pl.matisiek.pos.tracking.Main.main(Main.java:13)
Caused by: java.lang.ClassNotFoundException: com.github.sarxos.webcam.Webcam
        at java.net.URLClassLoader.findClass(Unknown Source)
        at java.lang.ClassLoader.loadClass(Unknown Source)
        at sun.misc.Launcher$AppClassLoader.loadClass(Unknown Source)
        at java.lang.ClassLoader.loadClass(Unknown Source)
        ... 1 more
``
And this is my script:

package pl.matisiek.pos.tracking;

import java.io.File;
import java.io.IOException;

import javax.imageio.ImageIO;

import com.github.sarxos.webcam.Webcam;

public class Main {

	public static void main(String args[]) {
		Webcam webcam = Webcam.getDefault();
		if (webcam != null) {
			System.out.println("Webcam: " + webcam.getName());
		} else {
			System.out.println("No webcam detected");
		}
		webcam.open();
		try {
			ImageIO.write(webcam.getImage(), "PNG", new File("hello-world.png"));
		} catch (IOException e) {
			e.printStackTrace();
		}
	}
	
}

What is the problem???

PS. Jakby co możesz pisać po polsku, bo masz ustawione na profilu, że jesteś z Polski

sarxos, 2016-12-30:
Hej @MatisiekPL,

Wygląda na to, że nie dodałeś JAR-a do classpath. Wyjątek ```NoClassDefFoundError``` występuje tylko w kilku przypadkach:
1) gdy JAR nie jest w classpath, 
2) gdy statyczny inicjalizer wywali błąd (w klasie ```Webcam``` takowy nie występuje), który nie jest łapany,
3) gdy plik JAR jest uszkodzony.

Obstawiam przypadek 1. W zależności jak uruchamiasz swój program - z IDE czy z linii poleceń powinieneś albo dodać JAR do classpath z IDE (różnie w zależności od pragramu, np Eclipse i Netbeans mają inaczej) lub przez opcję ```-cp``` lub ```-classpath``` gdy uruchamiasz z linii poleceń.

MatisiekPL, 2016-12-31:
[image]

To jest mój Build Path. Dodałem wszystkie jar'y które znalazłem w paczce. Właśnie mnie to dziwił, bo chyba wszystko dobrze zaimportowałem...

Maruti-Nandan, 2021-06-11:
@matisiekpl 
How did you fixed that, even I am facing the same problem.

seujorgenochurras, 2024-03-06:
I'm having the same issue too

closed 2024-04-05 by matisiekpl