sarxos/webcam-capture #528: No image available after turning camera off then turning back on.

heyJar, 2017-02-25:
Good day sarxos,
I have a problem with me project. When running the project, camera is working just fine. I have a button to turn on/off the camera, if turned on the numpad is enabled and if off, numpad is disabled and camera will be enabled, but after turning on back the camera, the panel displays "No image available" but it can still capture bar codes, it has just no display. Hope you can help me, thank you.
[image]
[image]

pavlovaleksei, 2017-03-03:
I was faced with the same problem
`//Class StartRecorder
package recorder;

import com.github.sarxos.webcam.Webcam;

/**
 * Created by a.pavlov on 03.03.2017.
 */
public class StartRecord {
    public static void main(String[] args) {
        Start st = new Start();
       // Webcam cam = Webcam.getDefault();
        for (int i = 0; i < 3; i++) {

            st.m();

        }

        System.out.println("RECORDING IS STOP!\nFILE SAVE!");
        while (true){

        }
    }
}

//class Start
package recorder;

import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamEvent;
import com.github.sarxos.webcam.WebcamLock;
import com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver;
import com.xuggle.xuggler.Xuggler;

import javax.swing.*;

/**
 * Created by a.pavlov on 01.03.2017.
 *
 * @version 1.2
 */
public class Start{

    public void m() {

        //Экземпляр класса Recorder
        Recorder record = new Recorder();

        //Указываем адрес сохранения видеофайла
        record.setFilePath("C:/Pavlov/Java/projects/projects_in_idea/media/test1/");
        //Указываем кол-во проходов цикла (1 проход = 1 сек.)
        record.setRecordTime(1);
        //Вызываем метод записи
        try {
            record.record();
        } catch (InterruptedException e) {
            e.printStackTrace();
        }

        //Выводим результаты работы в консоль
        System.out.println("===================================================================================");
        System.out.println("Запись успешно завершена, файл сохранен!");
        System.out.println("Общее кол-во проходов = " + record.count);
        System.out.println("===================================================================================");

return;
        //Вывод диалогового окна по завершению
        //JOptionPane.showMessageDialog(null, "Запись успешно завершена, файл сохранен!\nОбщее кол-во проходов = " + record.count);
    }
[... 265 lines cut ...]
        IAudioSamples samples = IAudioSamples.make(SAMPLE_RATE / fps, CHANNEL_COUNT, soundFormat);
        audioLine.start();
        byte[] data = new byte[(SAMPLE_RATE * BYTE_DEPTH * CHANNEL_COUNT) / fps];
        int sz = audioLine.read(data, 0, data.length);
        samples.put(data, 0, 0, sz);
        audioTime += (sz);

        samples.setComplete(true, sz / (BYTE_DEPTH * CHANNEL_COUNT), SAMPLE_RATE, CHANNEL_COUNT, soundFormat,
                audioTime / (BYTE_DEPTH * CHANNEL_COUNT));

        return samples;
    }

    /**
     * Метод окончания записи аудио
     */
    public static void closeAudioRec() {
        if (audioLine != null) {
            audioLine.close();
            audioLine = null;
        }
    }

}
`

pavlovaleksei, 2017-03-03:
win 7 x64
Thank you!