sarxos/webcam-capture #533: How to get a unique camera ID?

pavlovaleksei, 2017-03-28:
How to get a unique camera ID?
Thank you!

sarxos, 2017-03-29:
Hi @pavlovaleksei,

We should distinguish between ID given to UVC device by OS, and the ID stored in camera hardware. The first one (ID given by OS) is not always available and depends on the OS type and capture driver used. This is integer number, e.g. 0, 1, 2, which is allocated when you connect camera to USB port. The first one will be 0, next 1, and next one will be 2. The second ID, stored in camera hardware is not available at all and there is no way to get it.

However, if you need a way to identify webcam you should be able to use ```getName()``` to get unique name:

```java
for (Webcam w : Webcam.getWebcams()) {
  System.out.println(w.getName());
}
```

After you have name stored somewhere you can use ```Webcam.getWebcamByName(name)``` to find it. These names will be the same for the time when devices are connected to USB.

```java
String cameraName = "HD Webcam 1";
Webcam camera = Webcam.getWebcamByName(cameraName);
if (camera == null) {
  // oops, no camera with such a name can be found 
}
```

Depending on the operating systems webcam name is composed of:

* Camera Name + ID, e.g. _HD Camera 2_ or _Integrated Webcam 0_ (on Windows)
* Video Device name, e.g. _/dev/video0_ or _/dev/video2_ (on Linux, Mac)

And for IP camera driver, cameras do not have ID at all. In this case name is URL of MJPEG stream or address where camera should get JPEG frame.

closed 2017-03-29 by sarxos

pavlovaleksei, 2017-03-29:
Thanks for the answer! I'll use it!

justynb, 2017-05-17:
When I look at the webcam-capture-0.3.10.jar, in the Webcam class, there's no "getWebcamByName" method.

I'm looking to do the same thing as pavlovaleksei: if I have multiple cameras attached, I want to be able to choose one of the cameras (by comparing substrings, or whatever method).  I'm not a Java developer (or anything OO) so I'm sure it's probably something pretty obvious.

justynb, 2017-05-18:
I accomplished this by following
https://github.com/sarxos/webcam-capture/issues/481

ankitkat042, 2023-06-21:
> Hi @pavlovaleksei,
>
> We should distinguish between ID given to UVC device by OS, and the ID stored in camera hardware. The first one (ID given by OS) is not always available and depends on the OS [...]

I am using a 12-camera setup, and all cameras are from the same manufacturer. When I restart my Windows machines, they get assigned randomly(from 0 to 11), and each time I  have to manually check which camera is which one because their names are also the same. The one workaround I found is to rename them using USBtree and then try to figure out the index, but It's still not clear enough. I am trying to find a solution to this. Ping me if anyone also facing the same issue.