sarxos/webcam-capture #534: Create dedicated class to handle TP-Link NC200 IP Camera

sarxos, 2017-04-07:
This is a code we tested together with Dan Hagiiani to access his TP-Link NC200 IP Camera. NC200 (and probably NC220 as well) are somehow tricky:

* User interface is on port 80 and MJPEG stream is on port 8080,
* Password to access MJPEG needs to be BASE64-encoded.

Therefore it would be easier to have dedicated class to handle these.

This code needs to be refined:

https://pastebin.com/4CjxLhuQ

```java
import java.awt.Dimension;
import java.io.UnsupportedEncodingException;
import java.net.MalformedURLException;
import java.nio.charset.Charset;
import java.nio.charset.StandardCharsets;

import javax.swing.JFrame;

import org.apache.commons.codec.binary.Base64;

import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamPanel;
import com.github.sarxos.webcam.ds.ipcam.IpCamAuth;
import com.github.sarxos.webcam.ds.ipcam.IpCamDeviceRegistry;
import com.github.sarxos.webcam.ds.ipcam.IpCamDriver;
import com.github.sarxos.webcam.ds.ipcam.IpCamMode;

public class IpCameraTest {
	
	static {
		Webcam.setDriver(new IpCamDriver());
	}
	
	private static class AppWindow extends JFrame {

		/**
		 * Serial.
		 */
		private static final long serialVersionUID = 1L;
		
		public AppWindow() {
			
			setTitle("Demo app");
			setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);
			setResizable(false);
			setPreferredSize(new Dimension(640, 480));
			
			Webcam webcam = Webcam.getDefault();
			WebcamPanel panel = new WebcamPanel(webcam);
			
			
			
			add(panel);
			pack();
			setVisible(true);
		}
	}
	
	public static void main(String[] args) throws MalformedURLException {
		
		String name = "NC200";
		String url = "http://{ip-address}:8080/stream/getvideo"; 
	
		String user = "user";
		String password = encode("password");
		IpCamMode mode = IpCamMode.PUSH;
		IpCamAuth auth = new IpCamAuth(user, password);
	
		
		IpCamDeviceRegistry.register(name, url, mode, auth);
		
		new AppWindow();
		
	}

	private static String encode(String pwd) {
		final byte[] bytes = pwd.getBytes(StandardCharsets.UTF_8);
		final byte[] encoded = Base64.encodeBase64(bytes);
		try {
			return new String(encoded, "UTF-8");
		} catch (UnsupportedEncodingException e) {
			throw new IllegalStateException(e);
		}
	}
}
```

FuXual, 2017-08-06:
Hi! Thanks for a great piece of code 👍 :) I have tried it out and like the work and all about capturing camera. That's really nice! Now i would like to also get the microphone stream to be heard. Im not a Java developer so it's a bit harder to come up with a sulotion on that fast. I've searched around in your code and i can't find anything usefull to capture the audio. Have you plans on implementing this or do you have any tips for me how to proceed to get that working? Thanks in advance, and once again, thanks for some really nice piece of software! :)

sarxos, 2017-10-10:
Hi @FuXual, nope I do not plan to support microphone. If you are interested in webcam streaming with audio you may take a look at awesome job done by @ryancheu in his own fork:

https://github.com/ryancheu/webcam-capture

https://github.com/ryancheu/webcam-capture/commit/25d63d6c822a5aa3483e22c0ebb16f889276b666