sarxos/webcam-capture #535: package com.github.sarxos.webcam does not exist

siryabz, 2017-04-26:
Hi sir sarxos,

Is there someone had this error? I'm using netbeans 8.2 and i followed your instructions regarding the library and i found  this error. It says package does not exist but in the zip file its been there. Hope you can help me, thank you in advance.

[image]

closed 2017-05-01 by siryabz

artsakenos, 2017-08-05:
Same as above. How did you solve?
I'm also using Netbeans but I also tried to compile directly with mvn.
I also deleted the sarxos folder on M2 Home and tried again but with no success.

sarxos, 2017-08-05:
@Artsakenos

This is most likely caused because of the broken JAR file (e.g. broken
download, CRC error, etc). Just download file and verify if it's not
broken. Maven gives you control sum for that.

stvely, 2019-06-14:
I too am using NetBeans 8.2 and am trying to run your example "TakePictureExample.java". and I'm getting the same error.
run:
java.lang.ExceptionInInitializerError
Caused by: java.lang.RuntimeException: Uncompilable source code - package com.github.sarxos.webcam does not exist
	at TakePictureExample.<clinit>(TakePictureExample.java:8)
I was using webcam-capture-0.3.10.jar, slf4j-api-1.7.2.jar and bridj-0.6.2.jar but after reading your answer above I downloaded webcam-capture-0.3.12.jar, slf4j-api-1.7.2.jar and bridj-0.7.0.jar updated my class-path and even rebooted.
I have another program that I wrote that has been using webcam-capture-0.3.10.jar for some time without issue but I found out it has ceased to function so I tried your sample and found the error above.
I appreciate any help you can give.