sarxos/webcam-capture #536: Java VM Memory use getting higher when continous use without closing camera

j1mmyx, 2017-04-28:
Hi. thank you for library. I already use it over 2 years to create a simple home CCTV app for my own use.

I have a little problem here.

when continuous use (camera always open), Java VM Memory getting higher and higher, and then app crash because java low memory error (usually after 4-7 days). I don't know it's bugs from my app, Java VM, or your library.
I think I already check every single lines of my code and close all variable (or set to null) if not use anymore.

I have my own solution for my problem, I created thread to check memory use and automatic restart my app if memory use is higher than 85% for 20 sec.

-----------------------------------------------------------------------------------------

My app simple flows :
Thread 1
  - Getting image from your library
  - Check if there are motions (I created my own motion detector with auto tolerance to meet my home use needed)
  - Save image (if there are motion) to disk, in JPG or my own encrypted image format
Thread 2
  - Resize image to show it on bigger screen, I use JLabel to show image
Thread 3
  - Play notifications sound in 3 levels (low, medium and high motion) if there are motion
Thread 4
  - Restart app if FPS drop, threads errors, or low VM memory
Other Threads
  (No important things)

-----------------------------------------------------------------------------------------

My Spec for use app :
Netbook Lenovo s110
Intel Atom N2800 (2 Cores, 4 Threads) ~> Only get Max 5 FPS, but its ok for home CCTV
Ram 2 GB
USB Webcam (Logitech C525)

Last Software :
Windows 7 (32 bit)
Java 1.8.0_102 (32 bit),  Max memory is set to 247 MB (Default)

last library :
- bridj-0.6.2
- slf4j-api-1.7.2
- webcam-capture-0.3.10

-----------------------------------------------------------------------------------------

Note : 
please re-check your library if you have a time. This is not a big things, because your library is stable to run for long time, and 4 days running time in 247 MB java VM memory without closing camera is more than enough for me.

Sorry for my bad English.

sarxos, 2017-04-28:
Hi @j1mmyx,

Have you heard about a tool called jvisualvm? It's available once you install Oracle JDK. It's very helpful when it come to solving memory leaks. One of the fundamental tools of every good Java developer.

Here is a video on YouTube. You can jump straight to the 4:10 to see how he is using jvisualvm to inspect how many objects of a given class is allocated in the application. 

https://youtu.be/dUQqmnmCBbg?t=251

This is how heap inspector looks like:

[image]

In the same way you can check what is leaking in your app. The jvisualvm can be connected to application at every moment. It won't crash and won't cause failure. If this leak is caused by Java code then you will see it there. You can compare with application immediately after a launch to see what is happening. You can wait 3 days and then connect jvisualvm to check what objects are allocated an how many of them.

In the past I was running Webcam Capture for a weeks and haven't see any issue. Maybe something changed over the past few years and we need to re-evaluate.

j1mmyx, 2017-05-01:
Sorry for late reply, and thank you for your solution, I'm going to check that later. 
Your library is very helpful to connect any type of webcam to my application

sumsuddin, 2017-10-05:
I was having that problem too. Looks like [this](https://github.com/sarxos/webcam-capture/blob/64fdd35d9f44f2866997771db6b380771c510a5c/webcam-capture-drivers/driver-ipcam/src/main/java/com/github/sarxos/webcam/ds/ipcam/impl/IpCamMJPEGStream.java#L132) line is responsible for keeping the memory alive.

I think the memory would be released much faster if we could use a [try-resource-block](https://docs.oracle.com/javase/tutorial/essential/exceptions/tryResourceClose.html) for the `ByteArrayInputStream`.

However, my personal experience is that it's not an alarming issue. Because `GC` will eventually remove them from memory. So, if you have low memory then set the heap size lower such as pass -Xmx1024m