sarxos/webcam-capture #537: EXCEPTION_ACCESS_VIOLATION (0xc0000005) Problematic frame:  C  [ntdll.dll+0x50901]

pavlovaleksei, 2017-04-28:
Hello! I have a problem with cyclic video recording. The code is taken from the example: Encoder.
An error occurs when the loop passes more than 70 iterations.
Start-up example:

`package Recorder;

public class Start {
    public static void main(String[] args) {
        while(true) {
            try {
                Rec2 r = new Rec2();
                r.m();
                Thread.sleep(100);
            } catch (InterruptedException e) {
                e.printStackTrace();
            }
        }
    }
}

package Recorder;

import java.awt.Dimension;
import java.awt.image.BufferedImage;
import java.io.File;

import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamResolution;
import com.xuggle.mediatool.IMediaWriter;
import com.xuggle.mediatool.ToolFactory;
import com.xuggle.xuggler.ICodec;
import com.xuggle.xuggler.IPixelFormat;
import com.xuggle.xuggler.IVideoPicture;
import com.xuggle.xuggler.video.ConverterFactory;
import com.xuggle.xuggler.video.IConverter;

import java.util.Calendar;

public class Rec2 {

    public void m() {
        String filePath = new File("").getAbsolutePath() + "\\video\\";
        File file = new File(filePath + fileRenamed() + "mp4");

        IMediaWriter writer = ToolFactory.makeWriter(file.getName());
        Dimension size = WebcamResolution.QVGA.getSize();
        writer.addVideoStream(0, 0, ICodec.ID.CODEC_ID_H264, size.width, size.height);

        Webcam webcam = Webcam.getDefault();
        webcam.setViewSize(size);
        webcam.open(true);

        long start = System.currentTimeMillis();

        for (int i = 0; i < 50; i++) {

            System.out.println("Capture frame " + i);

            BufferedImage image = ConverterFactory.convertToType(webcam.getImage(), BufferedImage.TYPE_3BYTE_BGR);
            IConverter converter = ConverterFactory.createConverter(image, IPixelFormat.Type.YUV420P);

            IVideoPicture frame = converter.toPicture(image, (System.currentTimeMillis() - start) * 1000);
            frame.setKeyFrame(i == 0);
            frame.setQuality(0);

            writer.encodeVideo(0, frame);

            // 10 FPS
            try {
                Thread.sleep(100);
            } catch (InterruptedException e) {
                e.printStackTrace();
            }
        }

        writer.close();
        webcam.close();

        System.out.println("Video recorded in file: " + file.getAbsolutePath());
    }

    private String fileRenamed() {
        Calendar calendar = Calendar.getInstance();
        String fileName = String.valueOf(calendar.getTimeInMillis() + ".");
        return fileName;
    }
}`

pavlovaleksei, 2017-04-28:
Win 7 x64, Core i3-3210

sarxos, 2017-04-28:
Can you provide whole crash log?

pavlovaleksei, 2017-04-28:
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x0000000077580901, pid=7720, tid=0x0000000000000bf0
#
# JRE version: Java(TM) SE Runtime Environment (8.0_112-b15) (build 1.8.0_112-b15)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (25.112-b15 mixed mode windows-amd64 compressed oops)
# Problematic frame:
# C  [ntdll.dll+0x50901]
#
# Failed to write core dump. Minidumps are not enabled by default on client versions of Windows
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

---------------  T H R E A D  ---------------

Current thread (0x000000005500b000):  JavaThread "Finalizer" daemon [_thread_in_native, id=3056, stack(0x0000000056390000,0x0000000056490000)]

siginfo: ExceptionCode=0xc0000005, reading address 0x00000001d55734c8

Registers:
RAX=0x0000009d00001388, RBX=0x00000001d55734c0, RCX=0x000000ffffffffff, RDX=0x0000000000000000
RSP=0x000000005648f190, RBP=0x00000000586296e0, RSI=0x0000000000000000, RDI=0x0000000000000000
R8 =0x00000000586296d0, R9 =0x00000000c4e21668, R10=0x0000000002987f28, R11=0x000000005f9eb530
R12=0x0000000058620000, R13=0x0000000054daae58, R14=0x000000005648f350, R15=0x000000005500b000
RIP=0x0000000077580901, EFLAGS=0x0000000000010206

Top of Stack: (sp=0x000000005648f190)
0x000000005648f190:   000000005648f1a8 000000005648f1f8
0x000000005648f1a0:   0000000002977d80 00000000586296e0
0x000000005648f1b0:   0000000058628630 000000005648f1b8
0x000000005648f1c0:   00000000546d8814 000000005648f208
0x000000005648f1d0:   000000005500b000 000000005648f350
0x000000005648f1e0:   0000000000000000 0000000000000000
0x000000005648f1f0:   000000005d51aa31 000000005648f328
0x000000005648f200:   0000000054daae58 0000000076f51a7a
0x000000005648f210:   0000000054e2ee28 0000000054daae58
0x000000005648f220:   000000005648f2a0 0000000002978331
0x000000005648f230:   000000005648f2b0 000007fef2c1a750
0x000000005648f240:   00000000586296e0 000000005648f340
0x000000005648f250:   00000000ec450b20 00000000c4e23b60
0x000000005648f260:   0000000054daae58 000007fef2c17288
0x000000005648f270:   000000005648f2c0 00000000547ed7c0
0x000000005648f280:   0000000000000000 00000000546d8060 

Instructions: (pc=0x0000000077580901)
0x00000000775808e1:   ff ff 00 00 00 49 33 dc 48 23 c1 48 c1 eb 04 48
0x00000000775808f1:   33 d8 48 33 1d ce ea 0d 00 48 c1 e3 04 0f 0d 0b
0x0000000077580901:   48 8b 7b 08 49 bd 01 00 00 00 01 00 00 00 49 be
0x0000000077580911:   ff 7f 00 00 ff ff ff ff 41 c6 40 0f 80 41 88 70 

Register to memory mapping:

RAX=0x0000009d00001388 is an unknown value
RBX=0x00000001d55734c0 is an unknown value
RCX=0x000000ffffffffff is an unknown value
RDX=0x0000000000000000 is an unknown value
RSP=0x000000005648f190 is pointing into the stack for thread: 0x000000005500b000
RBP=0x00000000586296e0 is an unknown value
RSI=0x0000000000000000 is an unknown value
RDI=0x0000000000000000 is an unknown value
R8 =0x00000000586296d0 is an unknown value
R9 =0x00000000c4e21668 is an oop
java.lang.Class 
 - klass: 'java/lang/Class'
R10=0x0000000002987f28 is at code_begin+840 in an Interpreter codelet
[... 323 lines cut ...]
jvm_args: -javaagent:C:\Program Files (x86)\JetBrains\IntelliJ IDEA Community Edition 2017.1\lib\coverage-agent.jar=C:\Users\a.pavlov\AppData\Local\Temp\coverageargs -javaagent:C:\Program Files (x86)\JetBrains\IntelliJ IDEA Community Edition 2017.1\lib\idea_rt.jar=19270:C:\Program Files (x86)\JetBrains\IntelliJ IDEA Community Edition 2017.1\bin -Dfile.encoding=UTF-8 
java_command: Recorder.Start
java_class_path (initial): C:\Program Files\Java\jdk1.8.0_112\jre\lib\charsets.jar;C:\Program Files\Java\jdk1.8.0_112\jre\lib\deploy.jar;C:\Program Files\Java\jdk1.8.0_112\jre\lib\ext\access-bridge-64.jar;C:\Program Files\Java\jdk1.8.0_112\jre\lib\ext\cldrdata.jar;C:\Program Files\Java\jdk1.8.0_112\jre\lib\ext\dnsns.jar;C:\Program Files\Java\jdk1.8.0_112\jre\lib\ext\jaccess.jar;C:\Program Files\Java\jdk1.8.0_112\jre\lib\ext\jfxrt.jar;C:\Program Files\Java\jdk1.8.0_112\jre\lib\ext\localedata.jar;C:\Program Files\Java\jdk1.8.0_112\jre\lib\ext\nashorn.jar;C:\Program Files\Java\jdk1.8.0_112\jre\lib\ext\sunec.jar;C:\Program Files\Java\jdk1.8.0_112\jre\lib\ext\sunjce_provider.jar;C:\Program Files\Java\jdk1.8.0_112\jre\lib\ext\sunmscapi.jar;C:\Program Files\Java\jdk1.8.0_112\jre\lib\ext\sunpkcs11.jar;C:\Program Files\Java\jdk1.8.0_112\jre\lib\ext\zipfs.jar;C:\Program Files\Java\jdk1.8.0_112\jre\lib\javaws.jar;C:\Program Files\Java\jdk1.8.0_112\jre\lib\jce.jar;C:\Program Files\Java\jdk1.8.0_112\jre\lib\jfr.jar;C:\Program Files\Java\jdk1.8.0_112\jre\lib\jfxswt.jar;C:\Program Files\Java\jdk1.8.0_112\jre\lib\jsse.jar;C:\Program Files\Java\jdk1.8.0_112\jre\lib\management-agent.jar;C:\Program Files\Java\jdk1.8.0_112\jre\lib\plugin.jar;C:\Program Files\Java\jdk1.8.0_112\jre\lib\resources.jar;C:\Program Files\Java\jdk1.8.0_112\jre\lib\rt.jar;C:\Pavlov\Cloud_mail\farmnet\OnlyRecorder\out\production\OnlyRecorder;C:\Pavlov\Cloud_mail\farmnet\OnlyRecorder\lib_recorder\logback-classic-1.2.1.jar;C:\Pavlov\Cloud_mail\farmnet\OnlyRecorder\lib_recorder\bridj-0.7-20140918.jar;C:\Pavlov\Cloud_mail\farmnet\OnlyRecorder\lib_recorder\webcam-capture-0.3.12-20161206.184756-3.jar;C:\Pavlov\Cloud_mail\farmnet\OnlyRecorder\lib_recorder\commons-lang3-3.5.jar;C:\Pavlov\Cloud_mail\far [... cut ...]
CPU:total 4 (2 cores per cpu, 2 threads per core) family 6 model 58 stepping 9, cmov, cx8, fxsr, mmx, sse, sse2, sse3, ssse3, sse4.1, sse4.2, popcnt, avx, clmul, erms, ht, tsc, tscinvbit, tscinv

Memory: 4k page, physical 3881288k(303584k free), swap 8252260k(1420236k free)

vm_info: Java HotSpot(TM) 64-Bit Server VM (25.112-b15) for windows-amd64 JRE (1.8.0_112-b15), built on Sep 22 2016 21:31:56 by "java_re" with MS VC++ 10.0 (VS2010)

time: Fri Apr 28 14:36:04 2017
elapsed time: 384 seconds (0d 0h 6m 24s)

pavlovaleksei, 2017-04-28:
addition:
With recursive invocation (without a cycle while), the result is the same

sarxos, 2017-04-28:
Does the same happen when you get rid of Xuggler code and simply take pictures from camera in loop?

```
    public void m() {
        Dimension size = WebcamResolution.QVGA.getSize();

        Webcam webcam = Webcam.getDefault();
        webcam.setViewSize(size);
        webcam.open(true);

        for (int i = 0; i < 50; i++) {

            webcam.getImage();

            // 10 FPS
            try {
                Thread.sleep(100);
            } catch (InterruptedException e) {
                e.printStackTrace();
            }
        }

        webcam.close();
    }
```

pavlovaleksei, 2017-04-29:
Response. 
In this form code is working correctly. In a while loop passed 1000 iterations without crashing.

pavlovaleksei, 2017-05-02:
I think that the problem is in Bridj.jar

sarxos, 2017-05-08:
I actually think that Xuggler and BridJ natives interacts somehow and cause this crash to happen. If you are unable to work this out I suggest to move from Xuggler to [GStreamer Java bindings](https://github.com/gstreamer-java), but I have no examples on how to use it, however I'm certain it is possible to record video with gstreamer-java.

pavlovaleksei, 2017-05-10:
I'm still trying to figure this out.
There is a regularity when writing short (up to 5 seconds) files, the error flies to the 14th file, and when more long ones are written (about a minute), then about 340 files are recorded.

In addition, I found a way out by calling recursion and calling the garbage collector.
In this form, the code works almost constantly.
Below I give you an example of a code in which errors are almost not encountered.
Maybe my ideas will help you to improve the library. Thank you!

pavlovaleksei, 2017-05-10:
`public class Start {
    public static void main(String[] args) {
        rec r = new rec();
        r.m();
    }
}

import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamResolution;
import com.xuggle.mediatool.IMediaWriter;
import com.xuggle.mediatool.ToolFactory;
import com.xuggle.xuggler.ICodec;
import com.xuggle.xuggler.IPixelFormat;
import com.xuggle.xuggler.IVideoPicture;
import com.xuggle.xuggler.video.ConverterFactory;
import com.xuggle.xuggler.video.IConverter;

import java.awt.*;
import java.awt.image.BufferedImage;
import java.io.File;
import java.util.Calendar;

public class rec {

    BufferedImage image;
    IConverter converter;
    IVideoPicture frame;
    IMediaWriter writer;
    File file;
    String filePath;
    Webcam webcam;
    long start;
    Dimension size;
    
    public void m() {
        filePath = new File("").getAbsolutePath() + "\\video\\";
        file = new File(filePath + fileRenamed() + "mp4");

        writer = ToolFactory.makeWriter(file.getName());
        size = WebcamResolution.QVGA.getSize();
        writer.addVideoStream(0, 0, ICodec.ID.CODEC_ID_H264, size.width, size.height);

        webcam = Webcam.getDefault();
        webcam.setViewSize(size);
        webcam.open(true);

        start = System.currentTimeMillis();

        for (int i = 0; i < 50; i++) {

            System.out.println("Capture frame " + i);

            image = ConverterFactory.convertToType(webcam.getImage(), BufferedImage.TYPE_3BYTE_BGR);
            converter = ConverterFactory.createConverter(image, IPixelFormat.Type.YUV420P);

            frame = converter.toPicture(image, (System.currentTimeMillis() - start) * 1000);
            frame.setKeyFrame(i == 0);
            frame.setQuality(0);

            writer.encodeVideo(0, frame);

            // 10 FPS
            try {
                Thread.sleep(100);
            } catch (InterruptedException e) {
                e.printStackTrace();
            }
        }

        writer.close();
        webcam.close();
        
        filePath = null;
        file = null;
        writer = null;
        size = null;
        webcam = null;
        image = null;
        converter = null;
        frame = null;
        
        System.gc();
        
        System.out.println("Video recorded in file: " + file.getAbsolutePath());
        
        m();
    }

    private String fileRenamed() {
        Calendar calendar = Calendar.getInstance();
        String fileName = String.valueOf(calendar.getTimeInMillis() + ".");
        return fileName;
    }
}`

sarxos, 2017-05-10:
Hi, if GC is required maybe you should try to give more memory to your process? This can be done by specifying ```-Xmx``` flag in ```java``` command.

sarxos, 2017-05-10:
The default value of ```-Xmx``` is 64M which may be too low for video recording.

pavlovaleksei, 2017-05-10:
I pointed out and memory jvm settings, and settings for the stack, the situation has not changed radically, works more or less correctly only in this form

pavlovaleksei, 2017-05-10:
When installing in 512 MB and without calling the garbage collector, the process drops by OutOfMemory

sarxos, 2017-05-10:
Ok, thank you for the info.

pavlovaleksei, 2017-05-16:
This problem has been preserved for windows 10.
Can you tell me, is it possible to record many short video files (5 - 10 seconds) so that during the recording we connected to the camera 1 time, and while the camera was on, we wrote constantly short files?

kalilmvp, 2019-10-01:
Hello, i´ve beem having some issues also regarding the integration of my app and the webcam using this lib. I´m not sure if it´s the memory usage or some corrupt dll, but it´s also one time or another. I wanna get the crash log, but i haven´t seen this info anywhere, where can i find it ?

Thanks!

kalilmvp, 2019-10-01:
> This problem has been preserved for windows 10.
> Can you tell me, is it possible to record many short video files (5 - 10 seconds) so that during the recording we connected to the camera 1 time, and while the camera was on, we wrote constantly short files?

Hi 

> ```
> #
> # A fatal error has been detected by the Java Runtime Environment:
> #
> #  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x0000000077580901, pid=7720, tid=0x0000000000000bf0
> #
> # JRE version: [...]

Hi pavlo, i´m sorry if this is a dummy question, but how did you get the crash log ?