sarxos/webcam-capture #541: Maven don't Build

DuckHunter213, 2017-06-07:
Failed to execute goal on project webcam-capture-driver-gst1: Could not resolve dependencies for project com.github.sarxos:webcam-capture-driver-gst1:jar:0.3.12-SNAPSHOT: Failure to find org.freedesktop.gstreamer:gst1-java-core:jar:0.9-SNAPSHOT in http://sarxos.pl/maven2 was cached in the local repository, resolution will not be reattempted until the update interval of ftp-repo has elapsed or updates are forced -> [Help 1]

sarxos, 2017-08-07:
Ah, damn, I forget to add it to the repo ;( To w/a this you can disable webcam-capture-driver-gst1 module in the pom.

sarxos, 2017-10-10:
This is fixed in https://github.com/sarxos/webcam-capture/commit/e3e92c78c2f3594ff646cb36ab43cff350544f0a

closed 2017-10-10 by sarxos