sarxos/webcam-capture #544: Error: A JNI error has occurred, please check your installation and try again Exception in thread "main" java.lang.VerifyError: Cannot inherit from final class

suhagiyavishal111, 2017-07-12:
Hi @sarxos ,
first of all Thankyou so much for your effort and gave as such an amazing webcame functionality
I'm using Netbeans to develope my project. I have created a new project to test if the library function can scan QR code directly from webcam or not.

so i have used ur "WEBCAM-QR" project 
WebcamQRCodeExample.java
when i run it it gave error like:- 

Error: A JNI error has occurred, please check your installation and try again
Exception in thread "main" java.lang.VerifyError: Cannot inherit from final class
	at java.lang.ClassLoader.defineClass1(Native Method)
	at java.lang.ClassLoader.defineClass(ClassLoader.java:763)
	at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:142)
	at java.net.URLClassLoader.defineClass(URLClassLoader.java:467)
	at java.net.URLClassLoader.access$100(URLClassLoader.java:73)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:368)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:362)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(URLClassLoader.java:361)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:335)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	at java.lang.Class.getDeclaredMethods0(Native Method)
	at java.lang.Class.privateGetDeclaredMethods(Class.java:2701)
	at java.lang.Class.privateGetMethodRecursive(Class.java:3048)
	at java.lang.Class.getMethod0(Class.java:3018)
	at java.lang.Class.getMethod(Class.java:1784)
	at sun.launcher.LauncherHelper.validateMainClass(LauncherHelper.java:544)
	at sun.launcher.LauncherHelper.checkAndLoadMain(LauncherHelper.java:526)
/Users/peacock/Library/Caches/NetBeans/8.2/executor-snippets/run.xml:53: Java returned: 1
BUILD FAILED (total time: 1 second)

can you suggest solution 
Thankyou so much in advance.

sarxos, 2017-08-07:
I'm seeing this error for the first time ever. Do you have maybe some more details, especially the name of class that is final would be very helpful because this stack trace does not contain much details.

kg5000, 2017-09-25:
I am attempting to run a Lanterna demo program on an IBM Power8 AIX server with the IBM J9 JVM (see java -version below). 

This same error is occurring. 

bash-4.3$ java -cp textGui.jar:lanterna-3.0.0.jar demo.MultiButtonTest
Error: A JNI error has occurred, please check your installation and try again
Exception in thread "main" java.lang.NoClassDefFoundError: com.googlecode.lanterna.gui2.LayoutManager
        at java.lang.J9VMInternals.prepareClassImpl(Native Method)
        at java.lang.J9VMInternals.prepare(J9VMInternals.java:291)
        at java.lang.Class.getMethod(Class.java:1216)
        at sun.launcher.LauncherHelper.validateMainClass(LauncherHelper.java:556)
        at sun.launcher.LauncherHelper.checkAndLoadMain(LauncherHelper.java:538)
Caused by: java.lang.ClassNotFoundException: com.googlecode.lanterna.gui2.LayoutManager
        at java.net.URLClassLoader.findClass(URLClassLoader.java:609)
        at java.lang.ClassLoader.loadClassHelper(ClassLoader.java:850)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:829)
        at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:329)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:809)
        ... 5 more

[DemoCode.zip](https://github.com/sarxos/webcam-capture/files/1330372/DemoCode.zip)

bash-4.3$ java -version
java version "1.8.0"
Java(TM) SE Runtime Environment (build pap6480sr4fp6-20170518_02(SR4 FP6))
IBM J9 VM (build 2.8, JRE 1.8.0 AIX ppc64-64 Compressed References 20170516_348050 (JIT enabled, AOT enabled)
J9VM - R28_20170516_1905_B348050
JIT  - tr.r14.java_20170516_348050
GC   - R28_20170516_1905_B348050_CMPRSS
J9CL - 20170516_348050)
JCL - 20170516_01 based on Oracle jdk8u131-b11

erwrow, 2017-10-24:
@kg5000 if you are using the "webcam-capture-driver-native" will throw that error (at least for me) when you don't append the *.dll files, in my case it was solved by adding those.

sachin2517, 2018-01-18:
Install the following jar's and check. It worked for me.
(jcommander,jar & bsh-1.3.0.jar).

ishaaniwani, 2018-10-20:
I faced the same problem. I later found out the reason for the problem was that I had named the package "java" and that it was an illegal package name.

It seems you already have the problem worked out, but in the future, I suggest refactoring classes and packages.