sarxos/webcam-capture #545: Unexpected exit

rajesh-rt, 2017-07-27:
Hi,

I'm using IPCamera and have setup a motion detection listener to capture images from stream and upload them to AWS S3.

I'm using Maven dependencies as below:
<dependency>
	  <groupId>com.github.sarxos</groupId>
	  <artifactId>webcam-capture</artifactId>
	  <version>0.3.11</version>
</dependency>
<dependency>
	    <groupId>com.github.sarxos</groupId>
	    <artifactId>webcam-capture-driver-ipcam</artifactId>
	    <version>0.3.11</version>
</dependency>

My application is stopping unexpectedly after a few minutes with the following error:

Exception in thread webcam-panel-scheduled-executor-1
com.github.sarxos.webcam.WebcamException: Cannot download image
	at com.github.sarxos.webcam.ds.ipcam.IpCamDevice.getImagePullMode(IpCamDevice.java:400)
	at com.github.sarxos.webcam.ds.ipcam.IpCamDevice.getImage(IpCamDevice.java:318)
	at com.github.sarxos.webcam.ds.cgt.WebcamGetImageTask.handle(WebcamGetImageTask.java:43)
	at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:48)
	at com.github.sarxos.webcam.ds.cgt.WebcamGetImageTask.getImage(WebcamGetImageTask.java:26)
	at com.github.sarxos.webcam.Webcam.getImage(Webcam.java:668)
	at com.github.sarxos.webcam.WebcamPanel$ImageUpdater.update(WebcamPanel.java:568)
	at com.github.sarxos.webcam.WebcamPanel$ImageUpdater.run(WebcamPanel.java:547)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:308)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$301(ScheduledThreadPoolExecutor.java:180)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:294)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.net.SocketException: Connection reset
	at java.net.SocketInputStream.read(SocketInputStream.java:210)
	at java.net.SocketInputStream.read(SocketInputStream.java:141)
	at org.apache.http.impl.io.AbstractSessionInputBuffer.read(AbstractSessionInputBuffer.java:204)
	at org.apache.http.impl.io.ContentLengthInputStream.read(ContentLengthInputStream.java:182)
	at org.apache.http.conn.EofSensorInputStream.read(EofSensorInputStream.java:138)
	at javax.imageio.stream.FileCacheImageInputStream.readUntil(FileCacheImageInputStream.java:142)
	at javax.imageio.stream.FileCacheImageInputStream.read(FileCacheImageInputStream.java:187)
	at com.sun.imageio.plugins.jpeg.JPEGImageReader.readInputData(JPEGImageReader.java:312)
	at com.sun.imageio.plugins.jpeg.JPEGImageReader.readImage(Native Method)
	at com.sun.imageio.plugins.jpeg.JPEGImageReader.readInternal(JPEGImageReader.java:1236)
	at com.sun.imageio.plugins.jpeg.JPEGImageReader.read(JPEGImageReader.java:1039)
	at javax.imageio.ImageIO.read(ImageIO.java:1448)
	at javax.imageio.ImageIO.read(ImageIO.java:1352)
	at com.github.sarxos.webcam.ds.ipcam.IpCamDevice.getImagePullMode(IpCamDevice.java:391)
	... 14 more

Any help is much appreciated.

sarxos, 2017-08-07:
I think this is because in IpCamDevice connection is not renewed after error. This can be fixed, but I have no much time to work on this.

rajesh-rt, 2017-10-18:
sarxos, could you tell me how can I fix this? I'm using an ip camera(HIKVISION DS-2CD2420F-I(Audio) 2MP IR Cube IP Camera) connected over wifi. I'm facing this issue whenever I want to access video stream of an ip camera. Earlier I used Logitecth HD webcam and hadn't faced this issue.

sarxos, 2017-10-18:
Sure, you can change try/catch in IpCamDevice.getImagePullMode(..) and add catch for SocketException and instead of returning or throwing exception renew connection to the IP camera.

https://github.com/sarxos/webcam-capture/blob/master/webcam-capture-drivers/driver-ipcam/src/main/java/com/github/sarxos/webcam/ds/ipcam/IpCamDevice.java#L375

sarxos, 2017-10-18:
I will be happy to merge pull request for this if you fix it.

rajesh-rt, 2017-10-20:
Hi, I encountered the same issue. This time its happening right after capturing an image.
I'm using HIKVISION IR Network Cube Camera(DS-2CD2420F-I). 

INFO com.github.sarxos.webcam.ds.cgt.WebcamOpenTask - Opening webcam Amzur SDG
Capturing image...
Writing image to file...
[shutdown-hook-1] INFO com.github.sarxos.webcam.WebcamShutdownHook - Automatic Amzur SDG deallocation
[shutdown-hook-1] INFO com.github.sarxos.webcam.Webcam - Disposing webcam Amzur SDG

sarxos, 2017-10-20:
@rajesh-rt, I won't be fixing this because I have no access to IP camera I could test this issue with. Feel free to fix this on your own. Please let me know if description I provided in my comment above is not sufficient for you to start working on a solution.