sarxos/webcam-capture #553: Problem with Webcam Capture plugin

EgemenCatal, 2017-09-16:
Good day to everyone.
I'm a senior year physics student and i'm developing a project which involves ImageJ as a tool to take some pictures via webcam and process them. In order to do that i decided to use the Webcam Capture plugin.
The plugin launches fine but whenever i try to take a picture or use the live feed, it returns some sort of error that i can't understand. My knowledge of java language is almost next to none. The error is shown here; https://imgur.com/a/ZepkP
Any help is appreciated. Thank you

sarxos, 2017-09-16:
Can you tell from where you have SLF4J JAR file? Is it taken from ZIP bundle of Webcam Capture or you downloaded it manually from Internet?

The error states that method invoked on instance of Logger does not exist. This happen very often when incorrect JAR version is used. The correct SLF4J version from Webcam Capture is 1.7.2. Let me know in case you can't find it.

EgemenCatal, 2017-09-18:
slf4j-api-1.7.2.jar file was included in the Webcam Capture bundle and is installed correctly in the ImageJ directory.

EgemenCatal, 2017-09-21:
I've solved the problem! I had to reinstall Windows 10 for an unrelated issue and somehow the plugin is working. Thank you for your time.

closed 2017-09-21 by sarxos