sarxos/webcam-capture #556: Want to getImage() without webcam.open()

josephraj2k16, 2017-09-24:
hi sarxos,Iam using your webcam-capture in my play project and thanks for the code.I already opened my webcam feed using some html tags and look  like this 
[image]

now when I want to take a snap using this code:
val webcam = Webcam.getDefault() 
                  	    if (webcam != null) {
                            	  println("Webcam: " + webcam.getName())
                            	 
                              	webcam.open()
                  	    }
                       else {
                      	        println("No webcam detected");
                      	        val x1= "failure"
                      	        
                      	    }
                       ImageIO.write(webcam.getImage(), "PNG",new File("D:/folder1/1.png"))
                       val x="Image captured from Webcam"
It cant able to take snap because the webcam is already opened and here also im trying to open it again using "webcam.open()".When I delete the line "webcam.open()" it throws a error as "[IllegalArgumentException: image == null!]".Is there any way to take snap without using open() function??

sarxos, 2017-09-25:
Hi @josephraj2k16, if I'm not mistaken you should be able to do that with the following code:

```java
// this should do the trick, set when you application is started
static {
  Webcam.setAutoOpenMode(true);
}
```

And then:

```java
Webcam webcam = Webcam.getDefault();
if (webcam != null) {
  println("Webcam: " + webcam.getName())
  ImageIO.write(webcam.getImage(), "PNG",new File("D:/folder1/1.png"))
} else {
  System.out.println("No webcam detected");
}
```

josephraj2k16, 2017-09-25:
I did as u said,I had set the Auto open mode as true,and deleted the webcam.open(),now  it says that execution exception:[WebcamException: Cannot execute task] on this line =>ImageIO.write(webcam.getImage(), "PNG",new File("D:/folder1/1.png"))

[image]

sarxos, 2017-09-25:
Can you provide whole stack trace?

josephraj2k16, 2017-09-25:
here is the whole code:(scala doesnt has the keyword "static" so i took off static for autoopenmode()

val in =  hello.bindFromRequest
   in.fold(
                errors => { 
                  BadRequest(views.html.index(errors))
                },
                name => {
                     
                        Webcam.setAutoOpenMode(true);
                        val webcam = Webcam.getDefault 
                  	    if (webcam != null) {
                            	  println("Webcam: " + webcam.getName)
                            	 // webcam.open()
                	    }
                       else {
                      	        println("No webcam detected")        
                       }
                       ImageIO.write(webcam.getImage(), "PNG",new File("D:/folder1/"+name+".png"))
                       val x="Image captured from Webcam"
                       Ok(views.html.welcome(x))
                })
  }

josephraj2k16, 2017-09-25:
I have a breakthrough in this!!!!
I ran your code separately,i had a response like :[warn] c.g.s.w.d.b.WebcamDefaultDevice - Different size obtained vs requested - [176x144] vs [160x120]. Setting correct one. New size is [160x120]

Is this the reason why we cant have the solution??

sarxos, 2017-09-25:
@josephraj2k16, 

The response you are writing about is just a warning. It will not cause exception in your solution. The reason behind previous exception is different.

Can you try with a different approach and open webcam before it's served by bind request? Try this:

```scala
val webcam = Webcam.getDefault 
webcam.open();

val in =  hello.bindFromRequest
   in.fold(
                errors => { 
                  BadRequest(views.html.index(errors))
                },
                name => {
                  	    if (webcam != null) {
                            	  println("Webcam: " + webcam.getName)
                	    }
                       else {
                      	        println("No webcam detected")        
                       }
                       ImageIO.write(webcam.getImage(), "PNG",new File("D:/folder1/"+name+".png"))
                       val x="Image captured from Webcam"
                       Ok(views.html.welcome(x))
                })
  }
```

reginavaleria96, 2017-12-06:
I have similar question with similar case. My code:
	```
public void capture(String path) throws IOException {
		Webcam.setAutoOpenMode(true);
		Webcam webcam = Webcam.getDefault();
		if (webcam != null) {
			System.out.println("Webcam: " + webcam.getName());
			webcam.setViewSize(new Dimension(640, 480));
			// webcam.open();
			BufferedImage image = webcam.getImage();
			ImageIO.write(image, "JPG", new File(path));
			System.out.println("Captured & Saved");
			// webcam.close();
		} else {
			System.out.println("No webcam detected");
		}
	}
```

the stack trace:
```

com.github.sarxos.webcam.WebcamException: Cannot execute task
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.process(WebcamProcessor.java:72)
	at com.github.sarxos.webcam.WebcamProcessor.process(WebcamProcessor.java:140)
	at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:46)
	at com.github.sarxos.webcam.ds.cgt.WebcamOpenTask.open(WebcamOpenTask.java:20)
	at com.github.sarxos.webcam.Webcam.open(Webcam.java:271)
	at com.github.sarxos.webcam.Webcam.open(Webcam.java:232)
	at com.github.sarxos.webcam.Webcam.isReady(Webcam.java:745)
	at com.github.sarxos.webcam.Webcam.getImage(Webcam.java:613)
	at com.regina.thesis.tools.Camera.capture(Camera.java:21)
	at com.regina.thesis.servlet.Capture.doGet(Capture.java:41)
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:624)
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:731)
	at org.apache.catalina.core.ApplicationFilterChain.internalDoFilter(ApplicationFilterChain.java:303)
	at org.apache.catalina.core.ApplicationFilterChain.doFilter(ApplicationFilterChain.java:208)
	at org.apache.tomcat.websocket.server.WsFilter.doFilter(WsFilter.java:52)
	at org.apache.catalina.core.ApplicationFilterChain.internalDoFilter(ApplicationFilterChain.java:241)
	at org.apache.catalina.core.ApplicationFilterChain.doFilter(ApplicationFilterChain.java:208)
	at org.apache.catalina.core.StandardWrapperValve.invoke(StandardWrapperValve.java:218)
	at org.apache.catalina.core.StandardContextValve.invoke(StandardContextValve.java:110)
	at org.apache.catalina.authenticator.AuthenticatorBase.invoke(AuthenticatorBase.java:506)
	at org.apache.catalina.core.StandardHostValve.invoke(StandardHostValve.java:169)
	at org.apache.catalina.valves.ErrorReportValve.invoke(ErrorReportValve.java:103)
	at org.apache.catalina.valves.AccessLogValve.invoke(AccessLogValve.java:962)
	at org.apache.catalina.core.StandardEngineValve.invoke(StandardEngineValve.java:116)
	at org.apache.catalina.connector.CoyoteAdapter.service(CoyoteAdapter.java:445)
	at org.apache.coyote.http11.AbstractHttp11Processor.process(AbstractHttp11Processor.java:1115)
	at org.apache.coyote.AbstractProtocol$AbstractConnectionHandler.process(AbstractProtocol.java:637)
	at org.apache.tomcat.util.net.JIoEndpoint$SocketProcessor.run(JIoEndpoint.java:318)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at org.apache.tomcat.util.threads.TaskThread$WrappingRunnable.run(TaskThread.java:61)
	at java.lang.Thread.run(Unknown Source)
Caused by: com.github.sarxos.webcam.WebcamException: Cannot start native grabber!
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice.open(WebcamDefaultDevice.java:330)
	at com.github.sarxos.webcam.ds.cgt.WebcamOpenTask.handle(WebcamOpenTask.java:38)
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	... 1 more
```
My guess is that the library cannot detect the webcam as being open/turned on. Do you have any suggestion for my case?

sarxos, 2017-12-06:
Hi @reginavaleria96, I already addressed your issue in #594. Your problem is not caused by missing webcam.open().

stefan-reich, 2018-01-02:
So is it possible to use the same webcam in multiple processes? I thought you couldn't do that.