sarxos/webcam-capture #557: I need help about implementing the plugin.

EgemenCatal, 2017-09-27:
I'm trying to use a custom interface to take pictures and process them. I'm using the IP_Demo.java that comes with the ImageJ defaultly. How can i assign the "Grab and Return" command to Macro1? Thank you for your time.  

[IP_Demo.txt](https://github.com/sarxos/webcam-capture/files/1338490/IP_Demo.txt)

sarxos, 2017-09-27:
Hi @EgemenCatal, you should be asking in [ImageJ](https://github.com/imagej/imagej) project. I know that Webcam Capture API can be integrate in the ImageJ with the help of [this plugin](https://imagej.nih.gov/ij/plugins/webcam-capture/index.html) but I'm not familiar with it and cannot give you any hint on how this can be done.

EgemenCatal, 2017-09-27:
Thank you anyways 👍

closed 2017-09-27 by sarxos