sarxos/webcam-capture #559: Can't load library: /tmp/BridJExtractedLibraries978435834650156898/OpenIMAJGrabber.so   System.load("/tmp/BridJExtractedLibraries978435834650156898/OpenIMAJGrabber.so")

ashish012, 2017-10-02:
i working webcam sacanning in raspberry pi 3 ,, when i run program it gives :                                                       **Native code library failed to load.
java.lang.UnsatisfiedLinkError: Can't load library: /tmp/BridJExtractedLibraries978435834650156898/OpenIMAJGrabber.so** 

while i used _System.load("/tmp/BridJExtractedLibraries978435834650156898/OpenIMAJGrabber.so");_   but could not solved the problem  ,,, my maven is 

<?xml version="1.0" encoding="UTF-8"?>
<project xmlns="http://maven.apache.org/POM/4.0.0" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:schemaLocation="http://maven.apache.org/POM/4.0.0 http://maven.apache.org/xsd/maven-4.0.0.xsd">
    <modelVersion>4.0.0</modelVersion>
    <groupId>com.mycompany</groupId>
    <artifactId>ZxingWebCam</artifactId>
    <version>1.0-SNAPSHOT</version>
    <packaging>jar</packaging> 
    <dependencies>
        <dependency>
	    <groupId>com.google.zxing</groupId>
	    <artifactId>core</artifactId>
	    <version>3.2.1</version>
	</dependency>
	 
      <dependency>
    <groupId>com.github.sarxos</groupId>
    <artifactId>webcam-capture-driver-v4l4j</artifactId>
    <version>0.3.12-SNAPSHOT</version>
</dependency>
	<dependency>
	    <groupId>com.google.zxing</groupId>
	    <artifactId>javase</artifactId>
	    <version>3.2.1</version>
	</dependency> 
        <dependency>
   <groupId>org.openimaj</groupId>
   <artifactId>core-video-capture</artifactId>
   <version>1.3</version>
</dependency>
        <dependency>
    <groupId>com.github.sarxos</groupId>
    <artifactId>webcam-capture</artifactId>
    <version>0.3.12-SNAPSHOT</version>
</dependency> 

<dependency>
       <groupId>org.slf4j</groupId>
       <artifactId>slf4j-api</artifactId>
       <version>1.7.5</version>
   </dependency>
   <dependency>
       <groupId>org.slf4j</groupId>
       <artifactId>slf4j-log4j12</artifactId>
       <version>1.7.5</version>
   </dependency>

    </dependencies> 
   <repositories>  <repository>
	<id>Sonatype OSS Snapshot Repository</id>
	<url>http://oss.sonatype.org/content/repositories/snapshots</url>
</repository> 
 </repositories>

    <properties>
        <project.build.sourceEncoding>UTF-8</project.build.sourceEncoding>
        <maven.compiler.source>1.8</maven.compiler.source>
        <maven.compiler.target>1.8</maven.compiler.target>
    </properties>

josephraj2k16, 2017-10-04:
you have to load BridJ.jar file into the library path from the repository I think so.

ashish012, 2017-10-05:
**@josephraj2k16**  I used BriJ.jar in my dependencies , but not resolved.

sarxos, 2017-10-05:
Did you try running your application with debug options enabled and collecting the output logs? Just make sure you have logs configured. https://github.com/sarxos/webcam-capture/wiki/How-To-Enable-Debug-Logs

sarxos, 2017-10-05:
And can you provide details on which BridJ version are you using?

ashish012, 2017-10-06:
firstly thanks to **@sarxos** and  **@josephraj2k16** to showing interest on my problem ..
i added Bridj-0.7.0 version and bogback-core and logback-classic jar (but i didn't use environment variable _OPENIMAJ_GRABBER_VERBOSE=true_)  so
my latest pom is : 
<?xml version="1.0" encoding="UTF-8"?>
<project xmlns="http://maven.apache.org/POM/4.0.0" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:schemaLocation="http://maven.apache.org/POM/4.0.0 http://maven.apache.org/xsd/maven-4.0.0.xsd">
    <modelVersion>4.0.0</modelVersion>
    <groupId>com.mycompany</groupId>
    <artifactId>JupitorCam</artifactId>
    <version>1.0-SNAPSHOT</version>
    <packaging>jar</packaging> 
      <dependencies>
        <dependency>
	    <groupId>com.google.zxing</groupId>
	    <artifactId>core</artifactId>
	    <version>3.2.1</version>
	</dependency>
	 
      <dependency>
    <groupId>com.github.sarxos</groupId>
    <artifactId>webcam-capture-driver-v4l4j</artifactId>
    <version>0.3.12-SNAPSHOT</version>
</dependency>
	<dependency>
	    <groupId>com.google.zxing</groupId>
	    <artifactId>javase</artifactId>
	    <version>3.2.1</version>
	</dependency> 
        <dependency>
   <groupId>org.openimaj</groupId>
   <artifactId>core-video-capture</artifactId>
   <version>1.3</version>
</dependency>
        <dependency>
    <groupId>com.github.sarxos</groupId>
    <artifactId>webcam-capture</artifactId>
    <version>0.3.12-SNAPSHOT</version>
</dependency> 

<dependency>
       <groupId>org.slf4j</groupId>
       <artifactId>slf4j-api</artifactId>
       <version>1.7.25</version>
   </dependency>

<dependency>
    <groupId>ch.qos.logback</groupId>
    <artifactId>logback-classic</artifactId>
    <version>1.2.3</version>
    <scope>test</scope>
</dependency> 

<dependency>
    <groupId>ch.qos.logback</groupId>
    <artifactId>logback-core</artifactId>
    <version>1.2.3</version>
</dependency>

<dependency>
    <groupId>com.nativelibs4java</groupId>
    <artifactId>bridj</artifactId>
    <version>0.7.0</version>
</dependency>

    </dependencies> 
   <repositories>  <repository>
	<id>Sonatype OSS Snapshot Repository</id>
	<url>http://oss.sonatype.org/content/repositories/snapshots</url>
</repository> 
 
     
[... 312 lines cut ...]
	at org.bridj.BridJ.getNativeLibrary(BridJ.java:1072)
	at org.bridj.BridJ.getNativeLibrary(BridJ.java:1049)
	at org.bridj.BridJ.getNativeLibrary(BridJ.java:602)
	at org.bridj.cpp.CPPRuntime.newCPPInstance(CPPRuntime.java:771)
	... 13 more
Oct 06, 2017 6:14:33 PM org.bridj.BridJ log
INFO: Deleted temporary library file '/tmp/BridJExtractedLibraries2838597167179249625/libbridj.so'
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries2838597167179249625/OpenIMAJGrabber.so : libjpeg.so.8: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries2838597167179249625/OpenIMAJGrabber.so : libjpeg.so.8: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries2838597167179249625/OpenIMAJGrabber.so : libjpeg.so.8: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries2838597167179249625/OpenIMAJGrabber.so : libjpeg.so.8: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries2838597167179249625/OpenIMAJGrabber.so : libjpeg.so.8: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries2838597167179249625/OpenIMAJGrabber.so : libjpeg.so.8: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries2838597167179249625/OpenIMAJGrabber.so : libjpeg.so.8: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries2838597167179249625/OpenIMAJGrabber.so : libjpeg.so.8: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries2838597167179249625/OpenIMAJGrabber.so : libjpeg.so.8: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries2838597167179249625/OpenIMAJGrabber.so : libjpeg.so.8: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries2838597167179249625/OpenIMAJGrabber.so : libjpeg.so.8: cannot open shared object file: No such file or directory
------------------------------------------------------------------------
BUILD FAILURE
------------------------------------------------------------------------
Total time: 20.078s
Finished at: Fri Oct 06 18:14:33 UTC 2017
Final Memory: 7M/19M
------------------------------------------------------------------------

sarxos, 2017-10-07:
Hi @ashish012,

Install ```libjpeg8``` because in the log output you can clearly see:

> BridJ: dlopen error when loading /tmp/BridJExtractedLibraries2838597167179249625/OpenIMAJGrabber.so : libjpeg.so.8: cannot open shared object file: No such file or directory

Just do:

```plain
sudo apt-get install libjpeg8
```

Please let us know if this solved the problem.

ashish012, 2017-10-07:
Dear **@sarxos** sir 
After installing libjpeg8 my out put console looking like----->

SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
WLogC: Logback JARs are missing in classpath
------------------------------------------------------------------------
BUILD FAILURE
------------------------------------------------------------------------
Total time: 15.638s
Finished at: Sat Oct 07 18:35:20 UTC 2017
Final Memory: 7M/18M

**_and when remove libjpeg8 , i got previous exception in  out put console_** 

Please help me out sir ,,,,,,,,,, from past days i'm not sleeping well.

sarxos, 2017-10-07:
Hi @ashish012, 

If you take a look at the code, then you see that these logs:

```
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
WLogC: Logback JARs are missing in classpath
```

Are not errors but warnings. You can get rid of these warning simply by adding logback-classic into the classpath. You may think you already added it because it's in your pom.xml, but when you take a closer look, you will notice you added it in ```test``` scope (which means it's used only in tests).

Please also use Logback and SLF4J which was tested with Webcam Captuer API. You are using Logback 1.2.3 and SLF4J 1.7.25, which are pretty new, and in Webcam Capture we are using Logback 1.0.9 and SLF4J 1.7.2:

```xml
<dependency>
  <groupId>org.slf4j</groupId>
  <artifactId>slf4j-api</artifactId>
  <version>1.7.2</version>
</dependency>
<dependency>
  <groupId>ch.qos.logback</groupId>
  <artifactId>logback-classic</artifactId>
  <version>1.0.9</version>
</dependency>
```

Please remove test scope from your Logback dependency so it's used in runtime, not only in test.

That's in regards to these SLF4J and Logback warnings, but these warnings did not failed your build.

How did you start this application so you get BUILD FAILURE? Are you using mvn run?

ashish012, 2017-10-08:
yes i using mvn in netbeans  -- now i have changed as you said .. my output  is
---------------------------------------------------
cd /home/pi/NetBeansProjects/JupitorCam; JAVA_HOME=/usr/lib/jvm/jdk-8-oracle-arm32-vfp-hflt /home/pi/Documents/netbeans/java/maven/bin/mvn "-Dexec.args=-Dwebcam.debug=true -Dbridj.veryVerbose=true -Dbridj.quiet=false -Dbridj.logCalls=true -Dbridj.debug=true -classpath %classpath com.mycompany.jupitorcam.WebcamQRCodeExample" -Dexec.executable=/usr/lib/jvm/jdk-8-oracle-arm32-vfp-hflt/bin/java org.codehaus.mojo:exec-maven-plugin:1.2.1:exec
Running NetBeans Compile On Save execution. Phase execution is skipped and output directories of dependency projects (with Compile on Save turned on) will be used instead of their jar artifacts.
Scanning for projects...
                                                                        
------------------------------------------------------------------------
Building JupitorCam 1.0-SNAPSHOT
------------------------------------------------------------------------
Downloading: https://oss.sonatype.org/content/groups/public/com/nativelibs4java/bridj/0.6.3-SNAPSHOT/bridj-0.6.3-20130201.095358-5.jar/com/github/sarxos/webcam-capture-driver-v4l4j/0.3.12-SNAPSHOT/maven-metadata.xml

Downloading: https://oss.sonatype.org/content/groups/public/com/nativelibs4java/bridj/0.6.3-SNAPSHOT/bridj-0.6.3-20130201.095358-5.jar/com/github/sarxos/webcam-capture-drivers/0.3.12-SNAPSHOT/maven-metadata.xml

Downloading: https://oss.sonatype.org/content/groups/public/com/nativelibs4java/bridj/0.6.3-SNAPSHOT/bridj-0.6.3-20130201.095358-5.jar/com/github/sarxos/webcam-capture-parent/0.3.12-SNAPSHOT/maven-metadata.xml

Downloading: https://oss.sonatype.org/content/groups/public/com/nativelibs4java/bridj/0.6.3-SNAPSHOT/bridj-0.6.3-20130201.095358-5.jar/com/github/sarxos/webcam-capture/0.3.12-SNAPSHOT/maven-metadata.xml

--- exec-maven-plugin:1.2.1:exec (default-cli) @ JupitorCam ---
07:16:36.734 [main] INFO  com.github.sarxos.webcam.Webcam - WebcamDefaultDriver capture driver will be used
07:16:36.755 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
Oct 08, 2017 7:16:36 AM org.bridj.BridJ log
INFO: Embedded resource paths for library 'bridj': [org/bridj/v0_7_0/lib/linux_armhf/libbridj.so, org/bridj/v0_7_0/lib/linux_armhf/bridj.so, org/bridj/v0_7_0/lib/linux_armhf/libbridj.jnilib, org/bridj/lib/linux_armhf/libbridj.so, org/bridj/lib/linux_armhf/bridj.so, org/bridj/lib/linux_armhf/libbridj.jnilib, lib/linux_armhf/libbridj.so, lib/linux_armhf/bridj.so, lib/linux_armhf/libbridj.jnilib, libs/linux_armhf/libbridj.so, libs/linux_armhf/bridj.so, libs/linux_armhf/libbridj.jnilib]
Oct 08, 2017 7:16:36 AM org.bridj.BridJ log
INFO: Library resources for bridj: [org/bridj/v0_7_0/lib/linux_armhf/libbridj.so, org/bridj/v0_7_0/lib/linux_armhf/bridj.so, org/bridj/v0_7_0/lib/linux_armhf/libbridj.jnilib, org/bridj/lib/linux_armhf/libbridj.so, org/bridj/lib/linux_armhf/bridj.so, org/bridj/lib/linux_armhf/libbridj.jnilib, lib/linux_armhf/libbridj.so, lib/linux_armhf/bridj.so, lib/linux_armhf/libbridj.jnilib, libs/linux_armhf/libbridj.so, libs/linux_armhf/bridj.so, libs/linux_armhf/libbridj.jnilib]
Oct 08, 2017 7:16:37 AM org.bridj.BridJ log
INFO: Loading library /tmp/BridJExtractedLibraries430329628735312363/libbridj.so
Oct 08, 2017 7:16:37 AM org.bridj.BridJ log
INFO: Loaded library /tmp/BridJExtractedLibraries430329628735312363/libbridj.so
Oct 08, 2017 7:16:37 AM org.bridj.BridJ log
INFO: Calls logs enabled
Oct 08, 2017 7:16:37 AM org.bridj.BridJ log
INFO: Runtime for com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber : org.bridj.cpp.CPPRuntime
Oct 08, 2017 7:16:37 AM org.bridj.BridJ log
INFO: Registering type com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber
Oct 08, 2017 7:16:37 AM org.bridj.BridJ log
INFO: Library named 'OpenIMAJGrabber' is associated to file '{bridj=/tmp/BridJExtractedLibraries430329628735312363/libbridj.so}'
Oct 08, 2017 7:16:37 AM org.bridj.BridJ log
INFO: Environment var LD_LIBRARY_PATH = /usr/lib/jvm/jdk-8-oracle-arm32-vfp-hflt/bin/../jre/lib/amd64:/usr/lib/jvm/jdk-8-oracle-arm32-vfp-hflt/bin/../jre/lib/i386:
Oct 08, 2017 7:16:37 AM org.bridj.BridJ log
INFO: Environment var DYLD_LIBRARY_PATH = null
Oct 08, 2017 7:16:37 AM org.bridj.BridJ log
INFO: Environment var PATH = /usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
Oct 08, 2017 7:16:37 AM org.bridj.BridJ log
INFO: Property java.library.path = /usr/lib/jvm/jdk-8-oracle-arm32-vfp-hflt/bin/../jre/lib/amd64:/usr/lib/jvm/jdk-8-oracle-arm32-vfp-hflt/bin/../jre/lib/i386::/usr/java/packages/lib/arm:/lib:/usr/lib
Oct 08, 2017 7:16:37 AM org.bridj.BridJ log
INFO: Property sun.boot.library.path = /usr/lib/jvm/jdk-8-oracle-arm32-vfp-hflt/jre/lib/arm
Oct 08, 2017 7:16:37 AM org.bridj.BridJ log
INFO: Property gnu.classpath.boot.library.path = null
Oct 08, 2017 7:16:37 AM org.bridj.BridJ log
INFO: Looking for library 'OpenIMAJGrabber' in paths [., /usr/local/sbin, /usr/local/bin, /usr/sbin, /usr/bin, /sbin, /bin, /usr/local/games, /usr/games, /lib, /usr/lib, /usr/lib/jvm/jdk-8-oracle-arm32-vfp-hflt/jre/lib/arm, /usr/lib/jvm/jdk-8-oracle-arm32-vfp-hflt/jre/bin, /lib/arm-linux-gnueabihf, /usr/lib/arm-linux-gnueabihf, /usr/lib, /lib, /usr/local/lib]
Oct 08, 2017 7:16:37 AM org.bridj.BridJ log
[... cut ...]
Failed to execute goal org.codehaus.mojo:exec-maven-plugin:1.2.1:exec (default-cli) on project JupitorCam: Command execution failed. Process exited with an error: 139 (Exit value: 139) -> [Help 1]

To see the full stack trace of the errors, re-run Maven with the -e switch.
Re-run Maven using the -X switch to enable full debug logging.

For more information about the errors and possible solutions, please read the following articles:
[Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException

ashish012, 2017-10-09:
sir  yes i using mvn in netbeans  -- now i have changed as you said .. my output  is:   
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
--- exec-maven-plugin:1.2.1:exec (default-cli) @ JupitorCam ---
07:16:36.734 [main] INFO  com.github.sarxos.webcam.Webcam - WebcamDefaultDriver capture driver will be used
07:16:36.755 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
Oct 08, 2017 7:16:36 AM org.bridj.BridJ log
INFO: Embedded resource paths for library 'bridj': [org/bridj/v0_7_0/lib/linux_armhf/libbridj.so, org/bridj/v0_7_0/lib/linux_armhf/bridj.so, org/bridj/v0_7_0/lib/linux_armhf/libbridj.jnilib, org/bridj/lib/linux_armhf/libbridj.so, org/bridj/lib/linux_armhf/bridj.so, org/bridj/lib/linux_armhf/libbridj.jnilib, lib/linux_armhf/libbridj.so, lib/linux_armhf/bridj.so, lib/linux_armhf/libbridj.jnilib, libs/linux_armhf/libbridj.so, libs/linux_armhf/bridj.so, libs/linux_armhf/libbridj.jnilib]
Oct 08, 2017 7:16:36 AM org.bridj.BridJ log
INFO: Library resources for bridj: [org/bridj/v0_7_0/lib/linux_armhf/libbridj.so, org/bridj/v0_7_0/lib/linux_armhf/bridj.so, org/bridj/v0_7_0/lib/linux_armhf/libbridj.jnilib, org/bridj/lib/linux_armhf/libbridj.so, org/bridj/lib/linux_armhf/bridj.so, org/bridj/lib/linux_armhf/libbridj.jnilib, lib/linux_armhf/libbridj.so, lib/linux_armhf/bridj.so, lib/linux_armhf/libbridj.jnilib, libs/linux_armhf/libbridj.so, libs/linux_armhf/bridj.so, libs/linux_armhf/libbridj.jnilib]
Oct 08, 2017 7:16:37 AM org.bridj.BridJ log
INFO: Loading library /tmp/BridJExtractedLibraries430329628735312363/libbridj.so
Oct 08, 2017 7:16:37 AM org.bridj.BridJ log
INFO: Loaded library /tmp/BridJExtractedLibraries430329628735312363/libbridj.so
Oct 08, 2017 7:16:37 AM org.bridj.BridJ log
INFO: Calls logs enabled
Oct 08, 2017 7:16:37 AM org.bridj.BridJ log
INFO: Runtime for com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber : org.bridj.cpp.CPPRuntime
Oct 08, 2017 7:16:37 AM org.bridj.BridJ log
INFO: Registering type com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber
Oct 08, 2017 7:16:37 AM org.bridj.BridJ log
INFO: Library named 'OpenIMAJGrabber' is associated to file '{bridj=/tmp/BridJExtractedLibraries430329628735312363/libbridj.so}'
Oct 08, 2017 7:16:37 AM org.bridj.BridJ log
INFO: Environment var LD_LIBRARY_PATH = /usr/lib/jvm/jdk-8-oracle-arm32-vfp-hflt/bin/../jre/lib/amd64:/usr/lib/jvm/jdk-8-oracle-arm32-vfp-hflt/bin/../jre/lib/i386:
Oct 08, 2017 7:16:37 AM org.bridj.BridJ log
INFO: Environment var DYLD_LIBRARY_PATH = null
Oct 08, 2017 7:16:37 AM org.bridj.BridJ log
INFO: Environment var PATH = /usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
Oct 08, 2017 7:16:37 AM org.bridj.BridJ log
INFO: Property java.library.path = /usr/lib/jvm/jdk-8-oracle-arm32-vfp-hflt/bin/../jre/lib/amd64:/usr/lib/jvm/jdk-8-oracle-arm32-vfp-hflt/bin/../jre/lib/i386::/usr/java/packages/lib/arm:/lib:/usr/lib
Oct 08, 2017 7:16:37 AM org.bridj.BridJ log
INFO: Property sun.boot.library.path = /usr/lib/jvm/jdk-8-oracle-arm32-vfp-hflt/jre/lib/arm
Oct 08, 2017 7:16:37 AM org.bridj.BridJ log
INFO: Property gnu.classpath.boot.library.path = null
Oct 08, 2017 7:16:37 AM org.bridj.BridJ log
INFO: Looking for library 'OpenIMAJGrabber' in paths [., /usr/local/sbin, /usr/local/bin, /usr/sbin, /usr/bin, /sbin, /bin, /usr/local/games, /usr/games, /lib, /usr/lib, /usr/lib/jvm/jdk-8-oracle-arm32-vfp-hflt/jre/lib/arm, /usr/lib/jvm/jdk-8-oracle-arm32-vfp-hflt/jre/bin, /lib/arm-linux-gnueabihf, /usr/lib/arm-linux-gnueabihf, /usr/lib, /lib, /usr/local/lib]
Oct 08, 2017 7:16:37 AM org.bridj.BridJ log
INFO: Possible file names for library 'OpenIMAJGrabber' with name 'OpenIMAJGrabber': [libOpenIMAJGrabber.so, OpenIMAJGrabber.so, libOpenIMAJGrabber.jnilib]
Oct 08, 2017 7:16:37 AM org.bridj.BridJ log
INFO: Embedded resource paths for library 'OpenIMAJGrabber': [com/github/sarxos/webcam/ds/buildin/lib/linux_armhf/libOpenIMAJGrabber.so, com/github/sarxos/webcam/ds/buildin/lib/linux_armhf/OpenIMAJGrabber.so, com/github/sarxos/webcam/ds/buildin/lib/linux_armhf/libOpenIMAJGrabber.jnilib, org/bridj/v0_7_0/lib/linux_armhf/libOpenIMAJGrabber.so, org/bridj/v0_7_0/lib/linux_armhf/OpenIMAJGrabber.so, org/bridj/v0_7_0/lib/linux_armhf/libOpenIMAJGrabber.jnilib, org/bridj/lib/linux_armhf/libOpenIMAJGrabber.so, org/bridj/lib/linux_armhf/OpenIMAJGrabber.so, org/bridj/lib/linux_armhf/libOpenIMAJGrabber.jnilib, lib/linux_armhf/libOpenIMAJGrabber.so, lib/linux_armhf/OpenIMAJGrabber.so, lib/linux_armhf/libOpenIMAJGrabber.jnilib, libs/linux_armhf/libOpenIMAJGrabber.so, libs/linux_armhf/OpenIMAJGrabber.so, libs/linux_armhf/libOpenIMAJGrabber.jnilib]
Oct 08, 2017 7:16:37 AM org.bridj.BridJ log
INFO: Library resources for OpenIMAJGrabber: [com/github/sarxos/webcam/ds/buildin/lib/linux_armhf/libOpenIMAJGrabber.so, com/github/sarxos/webcam/ds/buildin/lib/linux_armhf/OpenIMAJGrabber.so, com/github/sarxos/webcam/ds/buildin/lib/linux_armhf/libOpenIMAJGrabber.jnilib, org/bridj/v0_7_0/lib/linux_armhf/libOpenIMAJGrabber.so, org/bridj/v0_7_0/lib/linux_armhf/OpenIMAJGrabber.so, org/bridj/v0_7_0/lib/linux_armhf/libOpenIMAJGrabber.jnilib, org/bridj/lib/linux_armhf/libOpenIMAJGrabber.so, org/bridj/lib/linux_armhf/OpenIMAJGrabber.so, org/bridj/lib/linux_armhf/libOpenIMAJGrabber.jnilib, lib/linux_armhf/libOpenIMAJGrabber.so, lib/linux_armhf/OpenIMAJGrabber.so, lib/linux_armhf/libOpenIMAJGrabber.jnilib, libs/linux_armhf/libOpenIMAJGrabber [... cut ...]
INFO: Runtime for org.bridj.dyncallbacks.PointerToVoid_0 : org.bridj.CRuntime
Oct 08, 2017 7:16:38 AM org.bridj.BridJ log
INFO: // size = 0, alignment = -1
struct PointerToVoid_0 { 
}
Oct 08, 2017 7:16:38 AM org.bridj.BridJ log
INFO: No fields found in org.bridj.dyncallbacks.PointerToVoid_0 (maybe they weren't declared as public ?)
------------------------------------------------------------------------
BUILD FAILURE
------------------------------------------------------------------------
Total time: 14.750s
Finished at: Sun Oct 08 07:16:38 UTC 2017
Final Memory: 7M/18M

sarxos, 2017-10-09:
There is nothing bad in this log (except BUILD FAILURE at the end). The only problems are from BridJ demangler, but these are just a debug statements and can be ignored.

Did some crash occurred because I can't see full log, it's just cut in one moment:

```
INFO: No fields found in org.bridj.dyncallbacks.PointerToVoid_0 (maybe they weren't declared as public ?)
```

More things to check:
* Do you have enough RAM free to start application? Is this on your PC or on RasPi. If this is on Pi then how much RAM do you have there?
* Is the basic sample (without zxing) work fine?
* Can you try running it from command line (not via mvn exec) and outside NetBeans?
* Can you try bridj-0.6.3-SNAPSHOT instead of 0.7.0?

ashish012, 2017-10-12:
hi @sarxos  
sorry for late reply i was busy in my office work
really you have a great experience for resolving problem 
its working for me when i replace  **bridj-0.6.3-SNAPSHOT instead of 0.7.0**
and its also working from netbeans.

Thank you  for resolving my problem.
God always help you :)

sarxos, 2017-10-13:
Great. It's then BridJ issue. I'm closing this ticket.

closed 2017-10-13 by sarxos