sarxos/webcam-capture #562: Error: log4j:WARN No appenders could be found for logger (com.github.sarxos.webcam.Webcam). log4j:WARN Please initialize the log4j system properly.

Deepanchelliah, 2017-10-11:


Deepanchelliah, 2017-10-11:
I was trying to read qr code using webcam, i was getting this error. can you please explain me y am i getting these error. 

https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-examples/webcam-capture-qrcode

sarxos, 2017-10-11:
This is not error. As stated in a message, this is WARN, which means warning. You get it because you configured your log4j incorrectly. What are your JARs in classpath? If you do not need logs then you can simply ignore this warning and live without it :)

sarxos, 2017-10-11:
The first result from Google:

https://stackoverflow.com/questions/1140358/how-to-initialize-log4j-properly

https://www.google.com/search?q=log4j%3AWARN+Please+initialize+the+log4j+system+properly.

Deepanchelliah, 2017-10-12:
Thank you so much :)

closed 2017-10-12 by sarxos