sarxos/webcam-capture #564: Need IP cam image as a byte array of input stream.

sumsuddin, 2017-10-15:
I can see that the `WebcamDefaultDevice` is a `BufferAccess` object. But `IpCamDevice` is not. I need the original byte[] of the image.

 Is it possible to add that feature to `IpCamDevice` or is there any tweak that helps me to get the byte[] directly? Because for whatever reason `BufferedImage` to byte[] conversion takes more than 100ms time for my webcam having 2048 * 1536 resolution.

I have posted a [question](https://stackoverflow.com/q/46688650/5330223) on StackOverflow to make the conversion faster but no one seemed to answer that.

sarxos, 2017-10-22:
Yes. It is posible to add BufferAccess to IpCamDevice. One would have to simply read directly from stream and return byte[] instead of using ImageIO.

sumsuddin, 2017-10-23:
If I make a custom class modifying the IpCamDevice, it becomes hard for me to wire up everything so that the library uses my customised class.
Do I need to build the whole library locally rather than using `maven` or `gradle` dependency for this library?

sarxos, 2017-10-23:
Yes, the best way to solve your issue is to download sources, modify IpCamDevice and build with Maven.

```plain
$ git clone git@github.com:sarxos/webcam-capture.git webcam-capture-parent
$ cd webcam-capture-parent
// change IpCamDevice to support your needs
$ mvn clean install
// recompiled jar will be in webcam-capture-drivers/driver-ipcam/target
```

sumsuddin, 2017-10-23:
Thank you very much for your time. Reopen the issue if you have any plan to add this feature.

closed 2017-10-23 by sumsuddin

sarxos, 2017-10-23:
Ok, @sumsuddin feel free to send pull request if you are willing to implement this on your own. I will be happy to check it :)