sarxos/webcam-capture #565: how can i select an specific driver

erwrow, 2017-10-20:
Hi, i'm have 2 questions, first, does the api autoselect the driver? in case that this doesn't happen, how can i select the driver? i'm ussing processing and this is the code that i'm ussing to do that:

```
import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamPanel;
import com.github.sarxos.webcam.WebcamResolution;
import java.awt.Dimension;
import java.awt.Image;
import java.awt.image.BufferedImage;

Webcam webcam;

void setup()
{
  size(640, 480);
  for (Webcam camlist : Webcam.getWebcams ())
  {
    println(camlist.getName());
    for(Dimension res : camlist.getViewSizes()) println(res.toString());
  }
  Dimension resolution = new Dimension(640, 480);
  webcam = Webcam.getDefault();
  webcam.setCustomViewSizes(new Dimension[] { resolution });
  
  //Webcam.setDriver(new IpCamDriver());
  //Webcam.setDriver(new FsWebcamDriver());
  
  Webcam.setDriver(new GStreamerDriver());
 
  //Webcam.setDriver(new OpenImajDriver());
  //Webcam.setDriver(new V4l4jDriver());
  //Webcam.setDriver(new JmfDriver());
  //Webcam.setDriver(new LtiCivilDriver());
  //Webcam.setDriver(new VlcjDriver());
  //Webcam.setDriver(new JavaCvDriver());
  //Webcam.setDriver(new FFmpegCliDriver());
  
  webcam.setViewSize(resolution);
  webcam.open(true);
}
void draw()
{
  BufferedImage cap = webcam.getImage();
  PImage video = new PImage(cap.getWidth(),cap.getHeight(),PConstants.ARGB);
  cap.getRGB(0, 0, video.width, video.height, video.pixels, 0, video.width);
  video.updatePixels();

  image(video, width/2, height/2);
}
```

when i select a driver the console throws me the following error "cannot find a class or type named: "GStreamerDriver"", i'm doing this because i need to use a webcam of a windows tablet (with windows 8)  and without selecting an specific driver i cant makeit work (to many errors are trhown)

sarxos, 2017-10-20:
No. It does not autoselect driver yet. I have plans to do that. You are doing this correctly. You simply didn't add required JARs into the classpath.

erwrow, 2017-10-23:
@sarxos thanks for the answer, i've tryed with some JARs with the same (or similar result), wich one should i use?

sarxos, 2017-10-23:
Check this repository for ZIP files of a specific drivers. These ZIP files contains a libs directory inside which contains all required JARs and the driver JAR is available in the ZIP main directory. 

http://repo.sarxos.pl/maven2/index.php?dir=com%2Fgithub%2Fsarxos%2F

For example, in this directory you can find ZIP file for [webcam-capture-driver-vlcj](https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers/driver-vlcj):

http://repo.sarxos.pl/maven2/index.php?dir=com%2Fgithub%2Fsarxos%2Fwebcam-capture-driver-vlcj%2F0.3.10%2F

However, I strongly recommend **not to manage dependencies (JARs) manually** but use one of the dependencies management tool, like Maven, Gradle, Ivy or something like that. This project relies strongly on Maven and every stable driver has its own dependency deployed in Maven Central Repository.

ZIPs and JARs are deployed to Maven Central from time to time, but not very often, so if you are looking for a newest version, the best would be to git clone this project onto your computer and build it with Maven, so you have all required JARs installed locally.

sarxos, 2017-10-23:
@erwrow and by the way - for windows tablet I strongly recommend using [webcam-capture-driver-native](https://github.com/frankpapenmeier/webcam-capture-driver-native) by Frank Papenmeier. Community members reported it's the best solution when you want to run Webcam Capture API on machine (PC or tablet) with modern Windows.

https://github.com/frankpapenmeier/webcam-capture-driver-native

erwrow, 2017-10-24:
@sarxos thanks for the suggestion, now the code is the following:

```
import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamPanel;
import com.github.sarxos.webcam.WebcamResolution;
import java.awt.Dimension;
import java.awt.Image;
import java.awt.image.BufferedImage;
Webcam webcam;
int i = 0;
void setup()
{
  size(640, 480);
  imageMode(CENTER);
  for (Webcam camlist : Webcam.getWebcams ())
  {
    println(camlist.getName());
    for(Dimension res : camlist.getViewSizes()) println(res.toString());
  }
  Dimension resolution = new Dimension(640, 480);
  webcam = Webcam.getDefault();
  webcam.setCustomViewSizes(new Dimension[] { resolution });
  Webcam.setDriver(new NativeWebcamDriver());
  webcam.setViewSize(resolution);
  webcam.open(true);
}
void draw()
{
  background(200);
  BufferedImage cap = webcam.getImage();
  PImage video = new PImage(cap.getWidth(),cap.getHeight(),PConstants.ARGB);
  cap.getRGB(0, 0, video.width, video.height, video.pixels, 0, video.width);
  video.updatePixels();
  image(video, width/2, height/2);
}
```
and throws me the exception: "java.lang.NullPointerException" on the line "PImage video = new PImage(cap.getWidth(),cap.getHeight(),PConstants.ARGB);" what can it be?

thank you very much for the help that you are giving me

sarxos, 2017-10-25:
The null pointer exception in this line may be most likely caused by null returned from getImage() method. Just ignore draw when cap == null.

sarxos, 2017-10-25:
And please **read examples and documentation**. You are setting driver in a wrong place. It should be instantiated before everything else. Now you are setting it in a middle of your code and it should be the very first operation you do. Immediately after main method is invoked.

closed 2017-10-30 by sarxos