sarxos/webcam-capture #566: webcam in remote server throws exception

josephraj2k16, 2017-10-21:
hi sarxos,i have deployed webcam remotely in cloud using heroku platform.When i run my code locally in localhost,there is no issue,but when i had deployed the same code to run in a remote server it throws that a library is not properly loaded.the error is:
"java.lang.RuntimeException: Library 'OpenIMAJGrabber' was not loaded successfully from file '/tmp/BridJExtractedLibraries4585895804822167921/OpenIMAJGrabber.so'"
You have already specified a solution for this in here(https://github.com/sarxos/webcam-capture/issues/116).But in my case the whole code is present remotely.Where should I change this library??

sarxos, 2017-10-21:
Please correct me if I'm wrong. Are you trying to capture images on server in Heroku cloud? Does it even has a webcam connected?

josephraj2k16, 2017-10-22:
yes i have deployed the webcam capture code on heroku and iam trying to access the webcam of the system whichever visits that particular site.

sarxos, 2017-10-22:
This is possible only when you deployed a web page to Heroku and this page has a Java Applet embedded. This way whenever user is accessing the page and agree to launch this specific applet, his webcam will be accessed. Are you using applet?

josephraj2k16, 2017-10-23:
i didnt use any java applet,I simply wrote the scala code(which contains your webcam api of getimage() and new file() etc..) and deployed them using heroku.Heroku maybe installed some dependencies for that....

sarxos, 2017-10-23:
This won't work. Your scala app will be trying to access webcam where it's running, that is a Heroku Dyno. However, since Heroku Dynos are built on some virtualization containers technology, there is no webcam and probably required dependencies are not installed as well. The bottom line is that you won't be able to open webcam of users who access app you deployed on Heroku because your code is running on Dyno and not on the user's OS. And the reason you are getting error is because Dynos are unable to access webcam because webcam is a real hardware in a real computer, and Dyno are not real computers but virtual containers.

josephraj2k16, 2017-10-24:
Thank you for the information sarxos,now I completely understand about dyno.But I have included a javascript code to open the webcam on a deployed webpage,and it retrives the users' webcam.How is that possible?

sarxos, 2017-10-24:
JavaScript is executed in a browser, that is on the user-end, and thus it's able to access hardware available on the computer which runs given browser. Your Java application, from the other hand, is executed on the server-side, and therefore is able to access server's hardware, not the users' one.

To make your Java application run on the user-end instead of the server-side you have two option:

1. Create Java Applet to run in user's browser and access his/her webcam. Please note, however, that Applets technology is outdated and almost dead now and I do not recommend using it to stream video from user's webcam.
2. Create Java Web Start application. This is a little bit similar to classic Java application you would write to run on desktop, but difference is that your JARs are deployed on web page and when user click on the button, the application will start, however not in the browser (as in case of Applets), but as a normal desktop application, like Excel or Word, for example.

In regards to your use-case, that is to get image from user's webcam when he access your web page deployed on Heroku, I strongly recommend using JavaScript for this. JavaScript is well suited to run in a browser, and Java, from the other hand, is well suited to be run as desktop or server application. Thus, you should use technology which is best suited for your purpose, that is a JavaScript, in case of webcam capturing code deployed into a web page.

josephraj2k16, 2017-10-24:
thanks for the illustration and your patience.I now clearly understood:-)

sarxos, 2017-10-24:
Cool, take care!

closed 2017-10-24 by sarxos